robotology/mesh-support #960: Icub neck encoder probably broken

S-Dafarra, 2020-07-14:
## Brief description of the request/failure
During a walking experiment the robot fell and the neck encoders went in hardware fault. The error is of the type
```
17036 5106.109870 <ERROR>  from BOARD 10.0.1.20 (head-eb20-j0_1), src LOCAL, adr 0, time 616s 429m 542u: (code 0x02000010, par16 0x0001 par64 0x0000000000000000) -> MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
```

## Detailed context

## Accompanying material

[log_icub-head_yarprobotinterface_4552_headEncoderError.txt](https://github.com/robotology/icub-tech-support/files/4919101/log_icub-head_yarprobotinterface_4552_headEncoderError.txt)

S-Dafarra, 2020-07-14:
cc @MiladShafiee

violadelbono, 2020-07-14:
The real problem was due to this error:

    14884 4591.055711 <ERROR>  from BOARD 10.0.1.20 (head-eb20-j0_1), src LOCAL, adr 0, time 101s 376m 562u: (code 0x02000008, par16 0x0001 par64 0x0000000000000009) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .

meaning that the encoder was not able to correctly read after the fall.

We checked the sensor and re-started the robot. Now it works correctly. I think we can close this for now and investigate more in case it happens again.

closed 2020-07-14 by violadelbono