robotology/mesh-support #961: Right arm wrist pitch and yaw joints does not run

MiladShafiee, 2020-07-14:
## Brief description of the request/failure
The right arm wrist pitch and yaw joint does not run and also the idle mode does not work(when I try to put in idle mode with motor gui)

## Detailed context
There is a hardware fault in right arm wrist pitch and yaw joints.

## Accompanying material

[log_icub-head_yarprobotinterface_1560-right-arm-wrist-pitch-yaw.txt](https://github.com/robotology/icub-tech-support/files/4920723/log_icub-head_yarprobotinterface_1560-right-arm-wrist-pitch-yaw.txt)

S-Dafarra, 2020-07-15:
The error seems related to #960. In addition, the neck joints go often in hardware fault with the same message of #960

julijenv, 2020-07-16:
It has been checked the movement after having found one screw missing, fixing the palm and checked the 2 AEA encoders, it seems ok to me, let me know

closed 2020-07-20 by julijenv