robotology/mesh-support #962: iCubGenova02 right leg calibration failure

Yeshasvitvs, 2020-07-15:
## Brief description of the request/failure

During some tests today, @lrapetti and I noticed the right leg calibration failure on iCubGenova02

## Detailed context

## Accompanying material

Terminal debug info

[right_leg_calibration_fail.txt](https://github.com/robotology/icub-tech-support/files/4926185/right_leg_calibration_fail.txt)

lrapetti, 2020-07-15:
The failure happened also when running the robot with the configuration launch file in `devel`

julijenv, 2020-07-16:
Hi @lrapetti , I just return from the purple place and check the problem, at the moment everything seems ok when just the yarprobotinterface is launched from robotSartup-eventdriven . Legs are stiff. To me it looks lie a config file error, could you double-check?

Yeshasvitvs, 2020-07-16:
@lrapetti we double checked this morning, and launching yarprobotinterface from the devel branch (which takes `icub_all.xml`) is running correctly without any failure of calibration of the right leg. Yesterday when you switched the branch to devel branch, did you install the robots-coniguration before running yarprobotinterface ?

lrapetti, 2020-07-16:
I was running the file directly from the folder inside `robotology-superbuild/robotology/robots-configuration` since I didn't want to touch the file inside the other `robots-configuration` folder that is located outside the `robtology-superbuild` (to which if I am not wrong the YARP_DATA_DIRS points directly, so there is no installation required)

 Anyway, yesterday I tried with `icub_wbd.xml` file, I am wondering if this can be the difference.

I can test again tomorrow to check if I experience the same behaviour.

lrapetti, 2020-07-17:
I tested it again today a few times with `icub_wbd.xml` and it started correctly all the time, I don't know what was wrong the last time but I think we can close the issue. Thanks.

closed 2020-07-17 by lrapetti