robotology/mesh-support #965: Cable of the left wrist broken

GiulioRomualdi, 2020-07-21:
##  Brief description of the request/failure
One of the cable of the left wrist is broken. 

cc @julijenv @S-Dafarra @violadelbono

lrapetti, 2020-07-21:
The wrist broke during the robot-robot experiments. In addiction we noticed that the left thumb was not started and this was happening even before the wrist failure. 

You can see here the status of the hand joints after the startup
[image]

Here you can find also the log of the `yarprobotinterface`: [log_icub-head_yarprobotinterface_1690.txt](https://github.com/robotology/icub-tech-support/files/4955984/log_icub-head_yarprobotinterface_1690.txt)

julijenv, 2020-07-22:
Hi @GiulioRomualdi  from the first estimation it seems that the coupling in between the motor and the transmission pulley is broken.
This will cost you a day of work adding to that the left thumb also broken and the right hand finger fingertips (#930 ) broken as well.
cc @violadelbono , @DanielePucci

julijenv, 2020-07-23:
Hi @GIuliRomuladi,
As expected the motor shaft is broken
we need to fetch from the warehouse the following pieces to restore

|IITCODE|Alias|version|Qty|Description|
|---|---|---|---|---|
|4471 | 1331T012SR_IE2-400_14-1 |  | 1| 1331T012SR-IE2-400+14/1-159:1+2082, Faulhaber - Brush motor, gear box, encoder with 500 mm cable length|

julijenv, 2020-07-23:
it has been used and will be to unload from warehouse

|IITCODE|Alias|version|S/N|Qty|Description|
|---|---|---|---|---|---|
|4471 | 1331T012SR_IE2-400_14-1 |  |1539 | 1| 1331T012SR-IE2-400+14/1-159:1+2082, Faulhaber - Brush motor, gear box, encoder with 500 mm cable length|
|2375|SHEATH | | |1| open spiral spring inox steel |

GiulioRomualdi, 2020-07-24:
Thank @julijenv and @violadelbono for handling the problem. The robot is now fixed and I think we can close the issue.

closed 2020-07-24 by violadelbono