robotology/mesh-support #966: Left shoulder cable broken

S-Dafarra, 2020-07-22:
## Brief description of the request/failure
While doing some tests for https://github.com/dic-iit/lab-events-demos/issues/146, one or two cables of the left shoulder broke

## Detailed context

## Accompanying material

cc @violadelbono @julijenv @kouroshD @Yeshasvitvs

violadelbono, 2020-07-22:
Hello @S-Dafarra,

The broken cable was # 8 and it has been replaced.
The following was taken from the warehouse:

| Item / IITCODE | Quantity |  Description | Notes  |
|:---:|:---:|:---|:---|
| 13737 | 1  | SHOULDER ROTATION CABLE 8, U7191215/SE000094 - 121mm - cyl. nipple AM420716/NI000007 - ball nipple KU000012 |  |

closed 2020-07-22 by violadelbono

DanielePucci, 2020-07-22:
The cable N.8 broke again. See https://github.com/dic-iit/lab-events-demos/issues/146#issuecomment-662563328

julijenv, 2020-07-23:
Hello @S-Dafarra, @DanielePucci ,

The broken cable was # 8 and it has been replaced one more time.
The following was taken from the warehouse:

| Item / IITCODE | ver|Quantity |  Description | Notes  |
|:---:|:---:|:---|:---|:---|
| 4765 |2| 1  | SHOULDER ROTATION CABLE 8 |  |