robotology/mesh-support #967: iCubGenova07 - Left shoulder cable 1 broken

dariopasquali, 2020-07-23:
Hi @julijenv and @violadelbono ,
we broke the left shoulder cable 1 of iCubGenova07 and we need a spare cable.
Also, cable 8 is damaged but we have a spare one.

It happened due to a sequence of failures:
- Cable 8 got stuck inside the shoulder joint
- Cables 1 and 2 move the block of pulleys of cable 8.
- So cable 1 is got pulled and broke while cable 2 just came out.

Also, the block of pulleys has a scratch in the inner circle.
[image]

------

Reading your [guide](https://icub-tech-iit.github.io/documentation/icub_tendons/shoulder_elbow/), I think we just need to remove the torso to access the crimps of cables 1 and 2.

However, we still need the spare cable and we have other open issues with both iCubGenova07 (#958) and iCubGenova03 (#953).
So I would ask you if you can please come here and fix the robots.

Thank you!
Dario

violadelbono, 2020-07-28:
Hello @dariopasquali and @reafrancesco,

The sequence of failure and the consequently scratch on the pulley was due to a defect in the fabrication of the latter, in particular the space for the crimp is not flat as it should be (see CAD model) but it has a step (see pic).

[image] [image]

Therefore, the cable crimp is not positioned perpendicularly but inclined, scratching the inner circle of the pulley and eventually let the cable jump when reaching a mechanical stop.

Cable 1 was replaced. We filed the crimp in order to match its seat as a temporary solution, but the pulley should be substituted/machined again.

dariopasquali, 2020-08-04:
Hi @julijenv and @violadelbono ,
We broke the cable 1 again, the very same sequence of failures:
- cable 1 is broken and we don't have the spare one
- cable 8 and cable 7 pulled out but no damage

We also have an update on issue #958 .

If you have time, could you fix it tomorrow?

violadelbono, 2020-08-07:
Hi @dariopasquali 

Yesterday I replaced cable 1 and fixed cables 8 an 7. The left shoulder is thus repaired.

However, in order to avoid further damage while waiting for fixing/replacing the pulley, we deactivated the left arm (see file icub_all.xml in iCubGenova07 folder).

violadelbono, 2020-09-11:
prova

violadelbono, 2020-10-15:
Hi @dariopasquali 

The shoulder pulley (IITCODE 3890.A1) was finally replaced on 13/10.
I close this issue.

dariopasquali, 2020-10-15:
Hi @violadelbono 
That's great! On the next week, we are also going to the shoulder.
For now, thanks for your work!

closed 2020-10-15 by dariopasquali

julijenv, 2020-10-15:
Hi @dariopasquali ,
we also recalibrate the right arm (the magnet was not glued as it should be - now it is), we didnt have time to copy the iCubGEnova0/calibrator/left_arm_calib.xml to your repository on github, could you please make a **diff** in between the one .local/etccc that we changed to recalibrate the elbo and the one $ROBOT_CODE/robotts-configuration/iCubGenova07 and just make sure that just the left_arm_claib.xml has been chenaged and later make a PR to have this change on github.
thx in advacne