robotology/mesh-support #969: Icub left camera's focal length error.

tKsome, 2020-07-31:
#

## Brief description of the request/failure

## Detailed context

## Accompanying material

/home/icub/Pictures/Screenshot from 2020-07-31 18-04-00.png

julijenv, 2020-07-31:
Hi @tKsome ,
Could you please try to fill up every single detail because from the time I was in Singapour the cameras were not that blurry.
Did you by any chance try to adjust by yourself the focus of the camera?
let me know.
BTW: IS it iCubSingapour01 the robot with which you are working on?

tKsome, 2020-07-31:
> Hi @tKsome ,
> Could you please try to fill up every single detail because from the time I was in Singapour the cameras were not that blurry.
> Did you by any chance try to adjust by yourself the focus of the camera?
> let me know.
> BTW: IS it iCubSingapour01 the robot with which you are working on?

sorry, i have added the detail.

tKsome, 2020-07-31:
> Hi @tKsome ,
> Could you please try to fill up every single detail because from the time I was in Singapour the cameras were not that blurry.
> Did you by any chance try to adjust by yourself the focus of the camera?
> let me know.
> BTW: IS it iCubSingapour01 the robot with which you are working on?

Do i need to take it off to adjust the focus?

pattacini, 2020-07-31:
@julijenv here's some more related context: https://github.com/robotology/icub-basic-demos/issues/25.

julijenv, 2020-07-31:
Hi @tKsome 
from

> Icub collapsed the other day after pressing the emergency stop button. After that, the image on the left camera becomes blurred, whether at 320x244 or 640x480 pixels.

I would love to know more about this falling, could you explain me if the robot was walking and fell down on the ground or if you use just the upper part of the body and the head smash some table or obstacle? If any of the following it looks like a normal push button for safety measure and should not have caused this behaviour.

To get this focal issue due to the way th camera is mounting 2 possibilitis for having this blurriness:
- the lens got unscrewed form its support (meaning screw holding the lens to the support got untight - red circle 
- or the support itself attached to the camera sensor got unscrewed (meaning screws - blue circles got untight)

[image]

could you check if by pushing or pressing the side of the lens, this brings back the correct image? or if it is when you try to tight or untighten the lens that the good image returns?
let us know

tKsome, 2020-08-01:
ICUB fell a few days ago while testing red-ball  tracking. It should be the reason that torso motor could not support the robot without power after pressing the stop button.
We're holding icub up on a platform, so it's just the upper body leaning forward.

Does it take icub apart to adjust the screws? Icub hasn't been disassembled yet.

julijenv, 2020-08-03:
Hi @tKsome ,
To assert that the camera is moving you dont need to unmount anything just try to compare both cameras trying to make them move inside the bulb. If you think that the lens is rotating (asserting that the red screw got released), just rotate it until you got better images, if you feel the lens steady but feel like the body of the lens is kind of moving so it means that the blue screws got lose. - to retighten the blue sccrews you will need to open the head cover nothing really frightening.
- open the head cover : remove the screw on each side of the head at the level of the ears.
 
[image]

- to retighten the red screw, 
[image]
unscrew the blue iris by holding firmly the bulb - once the iris is removed, retighten the screw with some tweezers.
[image]
let us know

julijenv, 2020-08-31:
Any news on this issue?
@tKsome

julijenv, 2020-12-30:
Hi @Ktsome,
did you succeed to put the focus back to normal on the left camera?
let us know and if yes please close the issue thx, if not please let me know.

closed 2021-02-18 by violadelbono