robotology/mesh-support #970: iCubErzelli01: yarprobotinterface doesn't start cause Bosh IMU fails

ddetommaso, 2020-08-06:
After updating PC104 software and boards firmware, yarprobotinterface stopped working.
The yarprobotinterface starts correctly when Inertial Sensors section is commented.

## Detailed context

**PC104 OS**
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.1 LTS
Release:	16.04
Codename:	xenial

**SW Distro:** 
2020.02.feat-01
## Accompanying material
[log_pc104_yarprobotinterface_4582.txt](https://github.com/robotology/icub-tech-support/files/5033472/log_pc104_yarprobotinterface_4582.txt)

julijenv, 2020-08-31:
Hi @ddetommaso , this isssue is still opened?
Unfortunately from the log file you sent me, i cant figure out what's going wrong. Let me know

ddetommaso, 2020-09-15:
Hi @julijenv 

please attached you can find a recent logfile using default icub_all.xml configuration.
[log_icub-head_yarprobotinterface_1823.txt](https://github.com/robotology/icub-tech-support/files/5223955/log_icub-head_yarprobotinterface_1823.txt)

julijenv, 2020-09-28:
Hi @ddetommaso ,

> 638 2.547453 <ERROR> 2 of 4: wrong eobrd_foc BOARD 10.0.1.6:CAN1:2 because it has:  WRONG APPLICATION VERSION  
> 639 2.547461 <ERROR> 4 of 4: wrong eobrd_foc BOARD 10.0.1.6:CAN1:4 because it has:  WRONG APPLICATION VERSION 

I guess you forgot to update the 2_FOC boards.
did you try again to update?

violadelbono, 2020-10-15:
- 2FOC firmware was updated
- Skin boards firmware still needs an update

@ddetommaso let us know if you have any issues after updating

ddetommaso, 2020-10-28:
Hi @violadelbono and @julijenv 

everything seems fine now! Thanks

julijenv, 2020-10-29:
hi @ddetommaso ,
even the first start error due to imu bosch???
if yes please close the issue, thx in advance

closed 2021-06-23 by github-actions[bot]