robotology/mesh-support #971: iCubErzelli01: yarprobotinterface with YARP_PORT_PREFIX causes AnalogServer Error initializing YARP ports

ddetommaso, 2020-08-06:
Setting YARP_PORT_PREFIX variable causes yarprobotinterface to not start the robot. The robot starts instead removing the Analog Sensors sections (MAIS and FT) from the configuration file.

## Detailed context

**SW Distro:**
2020.02.feat-01

## Accompanying material
[icub-head_yarprobotinterface_4905.txt](https://github.com/robotology/icub-tech-support/files/5033580/icub-head_yarprobotinterface_4905.txt)

julijenv, 2020-08-10:
Hi @ddetommaso ,
i've never seen this kind of error in my entire all life , COuld you please send me the icub_all_nolegs.xml , to chceck inside, thx in advance

pattacini, 2020-08-10:
@ddetommaso out of sheer curiosity, why did you launch `yarprobotinterface` with the `YARP_PORT_PREFIX ` env var? Was there any specific reason for that?

Just asking 'cause `yarprobotinterface` is quite at a low-level, whereas `YARP_PORT_PREFIX` is meant to manage high-level applications.

ddetommaso, 2020-08-24:
Hi @julijenv 

please in the following you can find the icub_all_nolegs.xml

`<?xml version="1.0" encoding="UTF-8" ?>
<!DOCTYPE robot PUBLIC "-//YARP//DTD yarprobotinterface 3.0//EN" "http://www.yarp.it/DTD/yarprobotinterfaceV3.0.dtd">

<robot name="iCubErzelli01" portprefix="icub" xmlns:xi="http://www.w3.org/2001/XInclude">
    <params>
    <xi:include href="hardware/electronics/pc104.xml" />
    </params>
        
    <devices>
    
    <xi:include href="./hardware/motorControl/head-eb20-j0_1-mc.xml" />  
    <xi:include href="./hardware/motorControl/head-eb21-j2_5-mc.xml" /> 
    <xi:include href="./wrappers/motorControl/head-mc_wrapper.xml" /> 
 
     
    <xi:include href="hardware/motorControl/torso-eb5-j0_2-mc.xml" />
    <xi:include href="wrappers/motorControl/torso-mc_wrapper.xml" /> 

    
    <xi:include href="hardware/motorControl/left_arm-eb1-j0_3-mc.xml" />
    <xi:include href="hardware/motorControl/left_arm-eb24-j4_7-mc.xml" />
    <xi:include href="hardware/motorControl/left_arm-eb25-j8_11-mc.xml" />
    <xi:include href="hardware/motorControl/left_arm-eb26-j12_15-mc.xml" />
    <xi:include href="wrappers/motorControl/left_arm-mc_wrapper.xml" /> 

    
    <xi:include href="hardware/motorControl/right_arm-eb3-j0_3-mc.xml" />
    <xi:include href="hardware/motorControl/right_arm-eb27-j4_7-mc.xml" />
    <xi:include href="hardware/motorControl/right_arm-eb28-j8_11-mc.xml" />
    <xi:include href="hardware/motorControl/right_arm-eb29-j12_15-mc.xml" />
    <xi:include href="wrappers/motorControl/right_arm-mc_wrapper.xml" />

    
    <xi:include href="wrappers/inertials/head-imuFilter_wrapper.xml" />
    <xi:include href="wrappers/inertials/head-imuFilter.xml" />
    <xi:include href="wrappers/inertials/head-inertials_wrapper.xml" />
    <xi:include href="wrappers/inertials/head-inertials_wrapper-deprecated.xml" />
    <xi:include href="hardware/inertials/head-inertial.xml" /> 

    
    <xi:include href="wrappers/MAIS/left_arm-mais_wrapper.xml" /> 
    <xi:include href="wrappers/MAIS/right_arm-mais_wrapper.xml" /> 
    <xi:include href="hardware/MAIS/left_arm-eb26-j12_15-mais.xml" /> 
    <xi:include href="hardware/MAIS/right_arm-eb29-j12_15-mais.xml" />  
    
     
    <xi:include href="wrappers/skin/left_arm-skin_wrapper.xml" />
    <xi:include href="hardware/skin/left_arm-eb24-j4_7-skin.xml" />
    <xi:include href="wrappers/skin/right_arm-skin_wrapper.xml" />  
    <xi:include href="hardware/skin/right_arm-eb27-j4_7-skin.xml" /> 
    <xi:include href="wrappers/skin/torso-skin_wrapper.xml" />
    <xi:include href="hardware/skin/torso-eb22-skin.xml" />
    
    
    <xi:include href="wrappers/FT/left_arm-FT_wrapper.xml" /> 
    <xi:include href="wrappers/FT/right_arm-FT_wrapper.xml" />
    <xi:include href="hardware/FT/left_arm-eb1-j0_3-strain.xml" /> 
    <xi:include href="hardware/FT/right_arm-eb3-j0_3-strain.xml" />

    
    <xi:include href="wrappers/VFT/left_arm-VFT_wrapper.xml" /> 
    <xi:include href="wrappers/VFT/right_arm-VFT_wrapper.xml" /> 
    <xi:include href="wrappers/VFT/torso-VFT_wrapper.xml" />
    
    
    
    <xi:include href="calibrators/left_arm-calib.xml" />
    <xi:include href="calibrators/right_arm-calib.xml" /> 
    <xi:include href="calibrators/torso-calib.xml" />
    <xi:include href="calibrators/head-calib.xml" />
   
     
    <xi:include href="cartesian/left_arm-cartesian.xml" />
    <xi:include href="cartesian/right_arm-cartesian.xml" /> 
    
    </devices>
</robot>`

ddetommaso, 2020-08-24:
> @ddetommaso out of sheer curiosity, why did you launch `yarprobotinterface` with the `YARP_PORT_PREFIX ` env var? Was there any specific reason for that?
> 
> Just asking 'cause `yarprobotinterface` is quite at a low-level, whereas `YARP_PORT_PREFIX` is meant to manage high-level applications.

@pattacini I would need to use two different robots with different names using the same YARP server. I decided for the YARP_PORT_PREFIX cause the --name param it seems obsolete in recent yarprobotinterface as reported [here](https://github.com/robotology/yarp/blame/1921d5041f8eb5bc73fa3453bb841678695452e3/src/yarprobotinterface/Module.cpp#L138). Please tell me if there is any alternative for customize icub name.

pattacini, 2020-08-25:
Never tried this myself @ddetommaso so I have no experience in your context, but some members of @robotology/iit-dynamic-interaction-control may chime in giving heads-up as they've been working out similar scenarios foreseeing robot-robot collaboration.

Yeshasvitvs, 2020-08-25:
@ddetommaso We use two real icub robots setup on the same network to perform robot-robot collaboration. We have a wiki page to on how to setup this scenarios, but at the moment it is a private document. I will share this document with you through teams. Eventually we will try to make this wiki public. 

CC @lrapetti @DanielePucci

pattacini, 2020-08-25:
Thanks heaps @Yeshasvitvs 👍🏻

ddetommaso, 2020-08-25:
Thanks a lot @pattacini and @Yeshasvitvs for the support!
I'll read carefully the solution you found and I'll try in my setup!

ddetommaso, 2020-08-25:
Hi @julijenv,

I close this issue cause I don't need further support.

@pattacini as general comment I think that it would be useful to address the issue of using multiple iCub robots within the same network (simulation and real scenario). In my opinion the easiest solution is to use a YARP port prefix as parameter of the yarprobotinterface. Then the yarprobotinterface should make the naming of the ports automatically based on this param or using /icub if YARP prefix is not set.

At the moment this is not possible cause the default YARP prefix (/icub) is hard-coded in some of the configuration files (see wrappers for example). This is why setting YARP_PORT_PREFIX in yarpmanager causes the AnalogServer Error. The workaround consists of renaming manually the configuration files where the prefix is hard-coded.

closed 2020-08-25 by ddetommaso

pattacini, 2020-08-25:
Thanks @ddetommaso for the update.

> @pattacini as general comment I think that it would be useful to address the issue of using multiple iCub robots within the same network (simulation and real scenario). In my opinion the easiest [...]

I'm afraid that this might be not that straightforward. I'm not an expert but, perhaps, the mechanism of `YARP_PORT_PREFIX` is not working in this context where `yarprobotinterface` is merely a deployer of dynamic plugins, which are actually responsible for opening up ports. I'm saying that `yarprobotinterface` might not be able to catch that a port gets opened up inside a plugin hence it's not possible to attach the prefix.

That said, I'm also concerned about the fact that other modules depending on those plugins (e.g. the Cartesian Solvers) need somehow to be customized to be aware of specific ports to connect to. This cannot be addressed by simply using `YARP_PORT_PREFIX` but rather by fiddling with configuration files anyway.

ddetommaso, 2020-08-26:
@pattacini thanks for your comments. I am not an expert too cause I had just a quick look to the code and I was wondering why the --name parameter was removed from yarprobotinterface. I see your point why it wouldn't be a straightforward solution modifying just yarprobotinterface. 

Again as a general comment, if the modules and plugins depends tightly to the configuration files then, maybe a generator of config files (configurator!?) might help with dynamic iCub names. Probably the config files are not only associated to a specific robot but mainly to its specific usage. As iCub user I can imagine that, depending on the specific use case, I might need a different configuration (network, hardware, yarp port names,...) output of such configurator. But I digress :D