robotology/mesh-support #972: iCubGenova04 - Left wrist yaw not controlled

lrapetti, 2020-08-07:
Yesterday during the robot-robot collaboration experiments the left wrist of the robot went to hardware fault. Restarting it, the yaw seems not to be controlled. The problem persist after restarting the robot, the wrist is not started, and when we try to start the wrist again, the yaw is not controlled.

## Accompanying material
Here is the log after restarting the robot: 
[log_icub-head_yarprobotinterface_2004.txt](https://github.com/robotology/icub-tech-support/files/5039934/log_icub-head_yarprobotinterface_2004.txt)

Yeshasvitvs, 2020-08-10:
@violadelbono @julijenv any update on this issue? We planned some experiments for today and tomorrow. So please let us know if we can fix this issue today. Thanks.

julijenv, 2020-08-11:
Ciao @Yeshasvitvs , and @lrapetti ,
the left wrist motor is broken (motor shaft) - we need a full day to restore it. as for now we are kind of busy behind a test ofr an upcoming robot getting out soon from production unit. We will right beck to you ASAP we got soon spare time.
we fixed the right wrist of iCubGenova02 as for now.

julijenv, 2020-09-28:
Hi guys as , you may have already seen the wrist has been repaired.
It has been made also an overall check and repaired 2 fingers

IITCODE | Qta | S/N | Description |  
-- | -- | -- | -- | --
4471 | 1 | 1537 | 1331T012SR-IE2-400+14/1-159:1+2082,   Faulhaber - Brush motor, gear box, encoder with 500 mm cable length |   |  
1875 | 1m |   | calrstahl 0.63mm non coated
2349 | 1 |   | right torsion spirng |

closed 2020-09-28 by julijenv