robotology/mesh-support #973: iCubGenova02 - right wrist broken

lrapetti, 2020-08-10:
The yaw cable of the robot right wrist seems to be broken/loose, as it can be observed in the following picture
[image]
also the following piece was found
[image]

cc @Yeshasvitvs

violadelbono, 2020-08-11:
Hi @lrapetti 

The following broken cable of the wrist was replaced:

| IITCODE  |  QUANTITY | ALIAS  | DESCRIPTION  |
|---|---|---|---|
| 6419  |1   | RC_IIT_019_G_028  | WRIST V2 TENDON # 10 - 45mm  |

lrapetti, 2020-08-11:
> Hi @lrapetti
> 
> The following broken cable of the wrist was replaced:

Great, thanks!
I think we can close the issue then.

lrapetti, 2020-08-11:
It seems the cable was not the only problem, the wrist is not starting even after changing the cable. There might be a problem similar to what observed on the green (https://github.com/robotology/icub-tech-support/issues/972).

marco-monforte, 2020-08-25:
Hi everyone! I confirm the motor shaft broken on the right side (like in the issue pointed by Lorenzo above) and I add the following issues on the left side:

- Calibration issue 

[image]

- Thumb proximal cable broke

[image]

- A pending capacitor from the back

[image]

violadelbono, 2020-09-04:
Yesterday the following cables of the left hand were fixed (and re-calibrated):

- Thumb proximal
- Ring and little fingers distal
- Middle distal

We also put a support for the capacitor.

List of items and quantity

| Item / IITCODE | Quantity |  Description |
|:---:|:---:|:---|
| 2035 . A2 | 1  | CAPACITOR HOLDING PLATE | 
| 1815 | 1m  | Stainless steel microcable, cable construction 7x7mm, nom. dia 0,63mm, not coated, AISI 316 | 
| 2391 | 1.5m | Stainless steel micro-cable, cable construction 7x19mm, nom. diameter 0.45mm, outer diameter 0.61mm, Polyamide coated, AISI 316 - CARL STAHL GMBH |

Nicogene, 2020-09-10:
I don't know id it is related to known issues of `iCubGenova02`, but today `gravityCompensator` crashed as soon as was connected to `wholeBodyDynamics`, both using the imu in the rfe and the classic imu in the head.

I ran it with gdb and I got:

[image]

cc @traversaro @pattacini

julijenv, 2020-09-28:
The left arm wrist motor was broekn and has been repaired.
afteer having remounted everything I didnt notice anything.

IITCODE | Qta | S/N | Description 
-- | -- | -- | -- 
4471 | 1 | 1537 | 1331T012SR-IE2-400+14/1-159:1+2082,   Faulhaber - Brush motor, gear box, encoder with 500 mm cable length

closed 2020-09-28 by julijenv