robotology/mesh-support #975: icubosaka: YARP network does not seem to be available, is the yarp server available?

YANGGANG1971, 2020-08-13:
After turning on icubsrv, icub01, icubVR2 and pc104,
run ./startYarpNetwork,sh on icubsrv.

When I type yarprun --server /icubo1 on icub01, there was something wrong as follows.
[image]

pattacini, 2020-08-13:
Hi @YANGGANG1971 

From the terminal where you launched `yarprun`, what will you get if you type the following command?
```sh
$ yarp where
```

Likely, you will be notified that the YARP server is not found. Thus, give a look at the machine whose IP is displayed in the output of the above command and see there if something wrong is happening.

In case you've changed the machine where YARP server is supposed to be running, then you'd need to update the remaining machines by doing on each:
```sh
$ yarp conf --clean
$ yarp detect --write
```
Alternatively, the following couple of commands is also doable:
```sh
$ yarp conf --clean
$ yarp conf <IP> <port>
```
where _IP_ accounts for the IP of the YARP server host and _port_ should be 10000.

Hope this helps out.

YANGGANG1971, 2020-08-13:
It works, thanks.

YANGGANG1971, 2020-08-13:
It works, thanks.

YANGGANG1971, 2020-08-13:
It works, thanks.

YANGGANG1971, 2020-08-13:
It works, thanks.

YANGGANG1971, 2020-08-13:
It works, thanks.

YANGGANG1971, 2020-08-13:
It works, thanks.

YANGGANG1971, 2020-08-13:
It works, thanks.

YANGGANG1971, 2020-08-13:
It works, thanks.

YANGGANG1971, 2020-08-13:
It works, thanks.

YANGGANG1971, 2020-08-13:
It works, thanks.

closed 2020-08-13 by pattacini