robotology/mesh-support #977: iCubShenzhen01: the tactile sensors on both thumbs are malfunctioning

tKsome, 2020-09-01:
## Detailed context
There was a problem with the tactile sensor of the thumb in the first few weeks. When the skin of the palm is touched, the tactile sensor of the thumb will always respond.

However, just now, I touched the palm of icub's left hand, and the tactile sensor of the thumb kept responding. Then I pressed the recalibration button and the tactile sensor of the thumb was reset.

After that, the tactile sensors of both thumbs were not working.

What is causing this? Is the wiring of the thumb's tactile sensor damaged?

julijenv, 2020-09-01:
Hi @tKsome ,
could you try to record a video of the problem occurring because I can not figure out your problem so much.
Do you mean that when you are touching the palm, you got some SPIKES on the visualization of the skin at the level of the thumb OR the thumb got ENTIRELY enlighted?
if it is just some spikes (for sure you have some grounding issues) you can get rid off them by clicking on the button ON-OFF of the binarization filter onto the QtSkinManagerGui window.
let me know.

tKsome, 2020-09-01:
[VID20200901201744.zip](https://github.com/robotology/icub-tech-support/files/5156115/VID20200901201744.zip)

tKsome, 2020-09-01:
_Do you mean that when you are touching the palm, you got some SPIKES on the visualization of the skin at the level of the thumb OR the thumb got ENTIRELY enlighted?_

Yes, BUT just sometime.

_if it is just some spikes (for sure you have some grounding issues) you can get rid off them by clicking on the button ON-OFF of the binarization filter onto the QtSkinManagerGui window._

I think this has nothing to do with the binary filter, it should be a problem with the touch sensor of the thumb. But I don't know where the problem occurred. What makes me more puzzled is that the same problem occurred on the other hand at the same time.

julijenv, 2020-09-01:
Hi @tKsome ,
could you please also record the problem in between the palm and thumb, because from your video it doesn't really match your interpretation of the situation?
thx in advance

violadelbono, 2020-09-01:
Hi @tKsome 

Looking at your video we also suggest to take a look at the thumb chip inside the back of the hand and check if it is correctly in position. 

First remove the back cover by unscrewing the M2x6 screw:
[image]

[image]

Let us know if this will solve the problem.

tKsome, 2020-09-02:
> Hi @tKsome ,
> could you please also record the problem in between the palm and thumb, because from your video it doesn't really match your interpretation of the situation?
> thx in advance

emmmm.... 
The reaction of the tactile sensor of the thumb after touching the palm only happens sometimes, and I cannot guarantee that it will not happen.

tKsome, 2020-09-02:
I opened the back cover and found the first chip protruding
[image]
[image]

violadelbono, 2020-09-02:
Hello @tKsome 

Is the problem still occurring after having pressed the chip in its position? It should now make a proper contact.

tKsome, 2020-09-06:
> Hello @tKsome
> 
> Is the problem still occurring after having pressed the chip in its position? It should now make a proper contact.

yes , it's work.

However, the back cover on the back of the hand will cause the chip to fall off. Can you send us the cad of the back cover? We redesign a more suitable back cover.

violadelbono, 2020-09-07:
Hello @tKsome 

I suggest you to clone our public github CAD repository :https://github.com/icub-tech-iit/cad-mechanics-public

Then go into this path: cad-mechanics-public\projects\icub\icub1_icub2 and look for the part named RC_IIT_017_P_030

tKsome, 2020-09-07:
> Hello @tKsome
> 
> I suggest you to clone our public github CAD repository :https://github.com/icub-tech-iit/cad-mechanics-public
> 
> Then go into this path: cad-mechanics-public\projects\icub\icub1_icub2 and look for the part named RC_IIT_017_P_030

OK, thanks.

closed 2020-10-08 by julijenv