robotology/mesh-support #978: Broken parts - iCubGenova01

xEnVrE, 2020-09-02:
## Brief description of the request/failure

I found some pieces of the robot on the desk, as depicted in the following

[image]

More in details:
- the finger nail is that of the right thumb finger(and is transparent instead of being white as all the others)
- the rubber ring is that of the left little finger
- the black piece belongs to the right leg (see figure below)

[image]

violadelbono, 2020-09-02:
@xEnVrE  I removed the rubber ring from the left thumb because I needed to tense the cable 😄 
I will replace it soon

xEnVrE, 2020-09-02:
> @xEnVrE I removed the rubber ring from the left thumb because I needed to tense the cable smile :smile:

@violadelbono sorry I didn't know it! I opened the issue just to avoid losing track of all these parts :smile:

violadelbono, 2020-09-02:
The robot is now fixed:

- I glued the broken piece of the cover
- replaced the nail 

| IIT CODE  | ALIAS  | QUANTITY  | DESCRIPTION  | 
|---|---|---|---|
|  6187 . A3  | RC_IIT_012_P_058  | 1  |  FINGERNAIL (left and right) |  

- Tensed proximal thumb joint and thumb oppose of both hands

-  replaced all the rubber rings missing

xEnVrE, 2020-09-02:
Thank you @violadelbono

closed 2020-09-02 by julijenv