robotology/mesh-support #979: iCubShenzhen01 - The thumb joint cable is loose

tKsome, 2020-09-11:
## Brief description of the request/failure
There was a collision between the thumb and the index finger, which caused the cable of the thumb joint9 to loosen when retracted.  The motor calibration of the right thumb cannot be completed after restarting.

## Accompanying material

[icub_thumb_cable.zip](https://github.com/robotology/icub-tech-support/files/5207808/icub_thumb_cable.zip)

tKsome, 2020-09-14:
hello???

The icub cable may be broken, but I am not sure if it is broken

pattacini, 2020-09-14:
Hi @tKsome 

Our mech support will be not available this week.
Sorry for the inconvenience.

tKsome, 2020-09-22:
> Hi @tKsome
> 
> Our mech support will be not available this week.
> Apologies for the inconvenience.

emmmm, 
Is the technical support department here this week？？？

maggia80, 2020-09-28:
Hi @tKsome, looking at the video it seems that the cable of the proximal joint of the thumb is broken. 
Instructions on how to fix it are at the following link: https://icub-tech-iit.github.io/documentation/icub_tendons/remove/

julijenv, 2020-09-28:
Hi @tKsome,
as already suspected by @maggia80 , the thumb proximal cable is quite loss - meaning 95% of the cases : cable broekn - 5% cases : just loose.
Could you please try to remove the cover of the forearm? and check the middle motor at the level of the elbow (see red circle - 
[image])
if the cable is broken you will see it from this place or just by pulling out the wire you may find out that the cable gets out.
Please follow the instructions from:
https://icub-tech-iit.github.io/documentation/icub_tendons/lower_arm_V2/#3-proximal-joints
Let us know

tKsome, 2020-09-30:
> Hi @tKsome,
> as already suspected by @maggia80 , the thumb proximal cable is quite loss - meaning 95% of the cases : cable broekn - 5% cases : just loose.
> Could you please try to remove the cover of [...]

but how can i open the tactile sensor in lower arm, it is tooooo complicated

julijenv, 2020-10-01:
Hi @tKsome ,
sorry I dont understand what you mean by :

> but how can i open the tactile sensor in lower arm, it is tooooo complicated

could you take a picture of what you are talking about?

BTW, I will be back in IIT on Monday, please contact me on skype:
my Id is icub.support
thx in advacne

tKsome, 2020-10-01:
Can we have a video meeting next Friday 9th Oct and fix the robot together? SZ is 7hours ahead. I think it is the best approach.

violadelbono, 2020-10-02:
Hi @tKsome,

We can schedule a Skype call for Oct Friday 9th at  9.30am CET which for you will be 3.30pm if I am correct.

Let me know!

tKsome, 2020-10-03:
> Hi @tKsome,
> 
> We can schedule a Skype call for Oct Friday 9th at 9.30am CET which for you will be 3.30pm if I am correct.
> 
> Let me know!

this time is OK. Looking forward to your support.

julijenv, 2020-10-08:
as it is a double issue let's close this one

closed 2020-10-08 by julijenv