robotology/mesh-support #980: Fix the hall effect position sensors of the right hand of iCubGenova04

kouroshD, 2020-09-11:
## Brief description of the request/failure
The hall effect position sensors of the right hand of the iCub is not streaming the raw analog sensors correctly.

## Detailed context
As currently the left wrist of the iCub Genova04 is broken https://github.com/robotology/icub-tech-support/issues/972 , I have only done the test with the right hand. But this is highly possible for the left hand as well. 

For retargeting the human hand motions to the robot hand motions, we rely on the hall effect sensors of the joints. But currently, the output of the `icub/right_hand/analog:o` port shows that, some sensors are not working correctly, and we do not have reliable outputs.
Here is the output of the aforementioned port for some instances:
```
255.0 240.0 255.0 255.0 118.0 197.0 93.0 234.0 249.0 0.0 0.0 0.0 237.0 163.0 255.0
255.0 240.0 255.0 255.0 118.0 197.0 93.0 234.0 249.0 0.0 0.0 0.0 237.0 163.0 255.0
```
In practice, as I have seen, the following finger sensors are not working:
- right thumb: the values of the sensors does not change coherently when moving the thumb in different directions using yarpMotorGui
- r_middle distal: the sensor is not places well, so from in the middle of motion it becomes zero and stay zeros, i.e., we can sense only a part of the moving range, the rest of it we have zero values readouts from the sensor.
- right ring finger :  all the values of it are zero.

Please, confirm if the rest of the sensors provides reasonable values for the whole range of motion of each joint of the finger, both for left and right hand.

## Accompanying material
```
255.0 240.0 255.0 255.0 118.0 197.0 93.0 234.0 249.0 0.0 0.0 0.0 237.0 163.0 255.0
255.0 240.0 255.0 255.0 118.0 197.0 93.0 234.0 249.0 0.0 0.0 0.0 237.0 163.0 255.0
255.0 240.0 255.0 255.0 118.0 197.0 93.0 234.0 249.0 0.0 0.0 0.0 237.0 163.0 255.0
255.0 240.0 255.0 255.0 118.0 197.0 93.0 234.0 249.0 0.0 0.0 0.0 237.0 163.0 255.0
255.0 240.0 255.0 255.0 118.0 197.0 93.0 234.0 249.0 0.0 0.0 0.0 237.0 163.0 255.0
255.0 240.0 255.0 255.0 118.0 197.0 93.0 234.0 249.0 0.0 0.0 0.0 237.0 163.0 255.0
255.0 240.0 255.0 255.0 118.0 197.0 93.0 234.0 249.0 0.0 0.0 0.0 237.0 163.0 255.0
255.0 240.0 255.0 255.0 118.0 197.0 93.0 234.0 249.0 0.0 0.0 0.0 237.0 163.0 255.0
255.0 240.0 255.0 255.0 118.0 197.0 93.0 234.0 249.0 0.0 0.0 0.0 237.0 163.0 255.0
255.0 240.0 255.0 255.0 118.0 197.0 93.0 234.0 249.0 0.0 0.0 0.0 237.0 163.0 255.0
Interrupting...
[INFO] |yarp.os.impl.PortCoreInputUnit| Removing input from /icub/right_hand/analog:o to /tmp/port/1
```

CC @DanielePucci

DanielePucci, 2020-09-14:
Hi guys, we would need support soon on this. I do not know how we could proceed @violadelbono @julijenv @Fabrizio69 @maggia80

maggia80, 2020-09-15:
I think that the issue can be a misconnection of the small boards on top of the MAIS board. Removing the top cover it's easy to see if this is the case. Let me know if you need help.

maggia80, 2020-09-15:
[image]

This is what I found.

kouroshD, 2020-09-16:
> This is what I found. There is one MAIS-FC connector disconnected from the MAIS for pourpose (there is tape around)

@maggia80  So, what is the plan for that?

maggia80, 2020-09-17:
we should wait @julijenv next week, because if it has been made for purpose, there should be a reason (I suspect a short circuit or something similar)

julijenv, 2020-09-28:
during the repair ofthe right arm #986, it seems that i resolve the problem

closed 2020-09-28 by julijenv