robotology/mesh-support #982: iCubErzelli01: Leftarm wrist not calibrating

ddetommaso, 2020-09-15:
## Brief description of the request/failure
The wrist of the left arm is not calibrating anymore.

## Accompanying material
[log_pc104_icubblue_yarprobotinterface_2143.txt](https://github.com/robotology/icub-tech-support/files/5224006/log_pc104_icubblue_yarprobotinterface_2143.txt)
[image]

julijenv, 2020-09-28:
Hi @ddetommaso ,
could you plese try to makea a video of the failing?
from the log file, the joint 7 seems out of boundries. Try to close the fingers and relaunch the robot and let me know

ddetommaso, 2020-09-30:
Hi @julijenv 

I've tried with different positions of the left arm but nothing changed.
Basically the wrist start moving during the calibration but it doesn't complete the movement.

If may help we already had a similar issue with this robot
https://github.com/robotology/icub-tech-support/issues/937

violadelbono, 2020-10-15:
Hi @ddetommaso 

We tensed the wrist tendons and re-calibrated the joint 7 (fingers abduction). As a result, the wrist now properly calibrate.
Issue can be closed.

closed 2020-10-15 by violadelbono