robotology/mesh-support #983: icubosaka01: icubsrv login loop

YANGGANG1971, 2020-09-15:
Followed by this tutorial, https://github.com/robotology/QA/issues/429, I git clone funny-things. It can work.
But when I restarted the icubsrv desktop and typed password, icubsrv were always in login interface. Maybe it is caused by operation incorrectly that we directly power off, not closed the terminals before. 

We have tried many methods (https://askubuntu.com/questions/223501/ubuntu-gets-stuck-in-a-login-loop), e.g. give the sudo permission to the specific file and change display management. 
Finally, it turned out that it does not work. 

Do you have any ideas for this?

YANGGANG1971, 2020-09-17:
Or how could I reinstall all software on icubsrv?

YANGGANG1971, 2020-09-25:
We have fixed the login issue. But if possible, we still intend to learn how to reset up icubsrv in case, also for backup.

mbrunettini, 2020-09-25:
@YANGGANG1971 : good news.
The installation instruxtions are available [here](http://wiki.icub.org/wiki/ICub_server_laptop_installation_instructions) : this guide has been written for Ubuntu 16.04, but it can be easily adapter to Ubuntu 18.04,

YANGGANG1971, 2020-09-25:
Thx, where can I download icub demo, like green ball, VR. yoga and emotion imitation.

pattacini, 2020-09-25:
>Thx, but where can I download icub demo, like green ball, VR. yoga and emotion imitation?

You may want to take a look here http://wiki.icub.org/wiki/Software_Versioning_Table.

pattacini, 2020-09-25:
Anyway @YANGGANG1971 

The instructions @mbrunettini gave you are regarded with the installation of a iCub laptop server.
Aside from the peculiarities of that installation, the code will get installed as for usual machines.

Look it up in our manual: http://wiki.icub.org/wiki/ICub_Software_Installation.

closed 2020-09-25 by pattacini

YANGGANG1971, 2020-09-25:
I mean the demo refers to the program after we first log in the icubsrv and then type yarpmanager. We will see many programs including many prerequisites, which help us judge its icub' status-good or bad. E.g. a motion imitation demo could help us judge icub's facial expression status. I thought before it is a set of default programs to test icub' status.

pattacini, 2020-09-25:
I got your point @YANGGANG1971 👍🏻 
Those instructions I mentioned above do cover most of the software you've seen on your laptop server.

Obviously, we're not responsible for the code that got installed therein and that do not belong to our traditional suite: for example, I'm not aware of any VR code we maintain.

YANGGANG1971, 2020-09-25:
Thank you all the same. Thx for your patient answer. Sorry about disturbing you always.

pattacini, 2020-09-25:
No problem @YANGGANG1971 
We're here to help 👍🏻