robotology/mesh-support #984: iCubGenova04 Battery pack not working

lrapetti, 2020-09-15:
One of the battery pack of robot `iCubGenova04` is not working. After it shut down (probably out of battery) we are no longer able to recharge it. 

What happens is that when we put it in charge, on the charger we observe that the CHARGE light (orange) is blinking as expected, however after short time the ALARM light (red) goes on. If we check the battery status pressing the button, the red light is always blinking.

CC @MrAndrea

lrapetti, 2020-09-15:
The charger we were using is the following one
[image]
[image]

MrAndrea, 2020-09-18:
I suppose that the battery pack is in undervoltage protection status due to the BMS, and the only way to come back to life is to connect opportunely and with extremely care the plus and minus poles to a power supply (42V, current limit eg 1A or 2A) .
I can check it next week.

S-Dafarra, 2020-09-30:
> I suppose that the battery pack is in undervoltage protection status due to the BMS, and the only way to come back to life is to connect opportunely and with extremely care the plus and minus poles to a power supply (42V, current limit eg 1A or 2A) .
> I can check it next week.
 
We tried to recharge it with the bigger blue power supplier. It seemed to work, but then the robot fell and we broke it definitely 😅

MrAndrea, 2020-12-28:
> We tried to recharge it with the bigger blue power supplier. It seemed to work, but then the robot fell and we broke it definitely 😅

I think we can close this issue, but if there are some pics of the broken battery pack please add them, it might be helpful to track the battery story and organise a possible repair.

S-Dafarra, 2021-01-04:
> > We tried to recharge it with the bigger blue power supplier. It seemed to work, but then the robot fell and we broke it definitely sweat_smile
> 
> I think we can close this issue, but if there are some pics of the broken battery pack please add them, it might be helpful to track the battery story and organise a possible repair.

I don't think we have it anymore in our open space. @DanielePucci brought it upstairs as soon as we unmounted it from the robot.

closed 2021-03-12 by violadelbono