robotology/mesh-support #985: iCubGenova04 left elbow cable out of track

lrapetti, 2020-09-16:
## Brief description of the request/failure

The cables from the left elbow appear to be out of the track, but they do not appear to be broken

[image]

## Detailed context
It happened while doing robot-robot collaboration experiments.

lrapetti, 2020-09-16:
Apparently @julijenv and @violadelbono are out of office this week, we should try to fix it because of upcoming experiments.

CC @DanielePucci @maggia80

lrapetti, 2020-09-16:
Update on the status, we tried to put the cable back on the track but the cable brake so it needs to be replaced.

cc @Giulero

maggia80, 2020-09-16:
https://icub-tech-iit.github.io/documentation/icub_tendons/shoulder_elbow/#3-left-elbow-cabling 

@Miche19  could you please provide them cable 9 and 10 (IITCODE 4766 e 4767)

Miche19, 2020-09-17:
@lrapetti the material is ready to be collected

cc: @maggia80

lrapetti, 2020-09-17:
Both the cables have been replaced and the elbow is now fixed.
Thanks @emilianob80 @prashanthr05 @Giulero

closed 2020-09-17 by lrapetti