robotology/mesh-support #986: Right wrist measurements are moving even if the arm is still

S-Dafarra, 2020-09-18:
## Brief description of the request/failure
The right wrist measurements jump even if the robot is perfectly still. I noticed this by looking at the iCub GUI
[image]

The robot was instead perfectly still.

## Detailed context
Here the log
[log_icub-head_yarprobotinterface_3097_right_arm_weird.txt](https://github.com/robotology/icub-tech-support/files/5246583/log_icub-head_yarprobotinterface_3097_right_arm_weird.txt)

cc @lrapetti @Yeshasvitvs @kouroshD @DanielePucci 

## Accompanying material

julijenv, 2020-09-28:
After having removed the right arm adn restore some of the problems on the last 2 fingers. I double check the connectgions ofthe AEA and recrimp 1 wire, hopefully this has restored the overall correct behaviour, I tried when I remounted everything and this should have gone.
please close the issue if yiu conrfimr that the problem dont show up.
@S-Dafarra

S-Dafarra, 2020-09-28:
> After having removed the right arm adn restore some of the problems on the last 2 fingers. I double check the connectgions ofthe AEA and recrimp 1 wire, hopefully this has restored the overall [...]

The arms have been removed again because of the iRonCub experiments, so I cannot check right now. In addition, if the problem appears again it may be a different problem. Hence, I prefer to close this and open a new one in case of problems. Thanks for dealing with this.

closed 2020-09-28 by S-Dafarra