robotology/mesh-support #987: Some cable related to the iCub's(icubgenova03) right hand's thumb could be broken

caampa, 2020-09-18:
## Brief description of the request/failure
Some cable related to the iCub's (**icubgenova03**) right hand's thumb could be broken

## Detailed context 
During the initial calibration, the thumb didn't move (we try it twice). Then, we try to move it using yarpmotorgui and we observed that the proximal and distal joints do not move. Still, the motor makes a sound, so probably a cable is loose. 

## Accompanying material

julijenv, 2020-09-28:
Hi @caampa ,
Could you please just record a short video to show the calibration of the finger and maybe also some of the movements you can make after calibration. It coudl help just to make sure that it is just a simple cable replcment

caampa, 2020-10-02:
[WhatsApp Video 2020-09-30 at 13.52.53.zip](https://github.com/robotology/icub-tech-support/files/5318370/WhatsApp.Video.2020-09-30.at.13.52.53.zip)
Here I attach a video recorder while the robot is being calibrated. 
Please ask me for more video if you need.
Thank you!

julijenv, 2020-10-05:
Hi @caampa ,
fomr the video your left thumb provximal joint looks broken.
I wll be in Erzelli on wednesday abd I will pass by to fix it up.

violadelbono, 2020-10-15:
Hello @caampa 

The right thumb proximal joint was repaired on 13/10.
The following items were taken from your toolbox:

|  IITCODE |  ALIAS  |   DESCRIPTION |   QUANTITY |
|---|---|---|---|
| 1815  | CG077063  | 1m  |  Stainless steel micro-cable, cable construction 7x7mm, nom. diameter 0,63mm, not coated, AISI 316 |  
|  2375 | SHEATH  | 9mm  | Open spiral spring, inox steel AISI302, D ext= 1.6mm, D wire= 0.25mm, L = 500 mm  |   

I was not able to launch the robot because of a problem with the pc104 (the server(s) could not ping the head) so when you will launch it next time please check that the finger works properly and then we can close the issue.

This problem could be due to the fact that the pc104 stayed ON for at least 3/4 days since on Tuesday we found it already ON. 
Let us know!

caampa, 2020-10-15:
Thank you very much @violadelbono!
I will keep you updated,

Elena

reafrancesco, 2020-11-09:
Thank you @julijenv @violadelbono. I managed to try the right hand`s thumb since you could not do it for the PC104 fail. I can confirm that the thumb cannot be controlled. For this reason I would wait to close this issue. Thanks.

julijenv, 2020-12-16:
It has been rewired and recalibrated both right and left hand with the help of @dariopasquali

closed 2020-12-16 by julijenv