robotology/mesh-support #989: Left arm wrist broken

lunagava, 2020-09-22:
## Brief description of the request/failure
Wrist yaw joint doesn't respond'  because wrist motor of left arm is broken. 
## Detailed context
Probably it happened because the hand put pressure on the air hockey table while iCub was grasping the paddle, even if I put limits on z direction. I'm very sorry about that. 

## Accompanying material

[image]

julijenv, 2020-09-28:
dear @lunagava ,
the motor of the wrist is broken, it will be repaired next week

lrapetti, 2020-10-16:
The problem is still there right?

I am asking because I have been using the robot without running wrist and fingers, and when I tested the whole-robot after my tests I have noticed the left wrist yaw seems to be still broken.

DanielePucci, 2020-10-16:
CC @julijenv @violadelbono

violadelbono, 2020-10-16:
Yes we didn't have time to fix it yet 😢

lrapetti, 2020-10-16:
> Yes we didn't have time to fix it yet 😢

No worries, I was just worried we broke it again :grimacing:
(we are currently running the robot without the wrist exactly to avoid that)

GiacomoVassallo13, 2020-11-16:
The crankshaft of the motor is broken into two parts, one of which is stuck in the pulley.
I try to extract it in the workshop, but I point out that this pulley has the hole that is visibly not aligned.
[image]
[image]

GiacomoVassallo13, 2020-11-16:
take the last motor (IIT cod. 4471) to the warehouse.
[image]

GiacomoVassallo13, 2020-11-16:
The pulley must be replaced , it must be replaced by taking the mechanical part from another kit, I await instructions on thisbecause it was not possible to recover it due to the deformation of the hole which led it to have a diameter of the hole for housing the shaft more fat already diameter of one millimeter.
the part number is RC_TLR_010_P_036
It must be replaced by taking the mechanical part from another kit, I await instructions on this
[image]

GiacomoVassallo13, 2020-11-17:
the broken motor was replaced with a working one and tested .
[image]

GiacomoVassallo13, 2020-11-17:
I noticed that the prosimal of the thumb needs to be rewired, I arrange it before reassembling the hand.
[image]

violadelbono, 2020-11-18:
After replacing the motor, the left arm was re-activated and now the robot is fixed.

I close the issue 🥳

closed 2020-11-18 by violadelbono