robotology/mesh-support #990: Multiple cables got broken after a fall

S-Dafarra, 2020-09-23:
## Brief description of the request/failure
During some shooting without crane, the robot fell hard on its back. One or more neck tendons and the right elbow tendon got broken.

## Detailed context
None

## Accompanying material

None

cc @DanielePucci @kouroshD @prashanthr05

S-Dafarra, 2020-09-25:
We received the robot back. Some update is also available at https://github.com/robotology/icub-tech-support/issues/673#issuecomment-698330558

Today @kouroshD tested it with the cable. He noticed that after a while the head simply shut down. This happened also before, but we were thinking about a power problem. We found it weird it happened also with the cable. By running ``journalctl -n 10000 --no-pager`` on a terminal on the head, there are these lines
```
Sep 25 09:17:01 icub-head CRON[3080]: pam_unix(cron:session): session closed for user root
Sep 25 09:43:54 icub-head wpa_supplicant[863]: wlp2s0: WPA: Group rekeying completed with 60:38:e0:da:4c:18 [GTK=CCMP]
Sep 25 09:45:32 icub-head kernel: perf: interrupt took too long (3145 > 3137), lowering kernel.perf_event_max_sample_rate to 63000
Sep 25 10:17:01 icub-head CRON[3273]: pam_unix(cron:session): session opened for user root by (uid=0)
Sep 25 10:17:01 icub-head CRON[3274]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Sep 25 10:17:01 icub-head CRON[3273]: pam_unix(cron:session): session closed for user root
-- Reboot --
Sep 25 10:25:12 icub-head kernel: microcode: microcode updated early to revision 0xd6, date = 2020-04-27
Sep 25 10:25:12 icub-head kernel: Linux version 4.15.0-109-lowlatency (buildd@lgw01-amd64-010) (gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04)) #110-Ubuntu SMP PREEMPT Tue Jun 23 03:19:37 UTC 2020 (Ubuntu 4.15.0-109.110-lowlatency 4.15.18)
Sep 25 10:25:12 icub-head kernel: Command line: BOOT_IMAGE=/boot/vmlinuz-4.15.0-109-lowlatency root=UUID=209ab836-eefa-11e8-b8c9-00e04b6543ae ro
Sep 25 10:25:12 icub-head kernel: KERNEL supported cpus:
Sep 25 10:25:12 icub-head kernel:   Intel GenuineIntel
Sep 25 10:25:12 icub-head kernel:   AMD AuthenticAMD
Sep 25 10:25:12 icub-head kernel:   Centaur CentaurHauls
```
Apart from ``-- Reboot --`` there seems to be no indication of what it could have happened.

Here the complete output of ``journalctl -n 10000 --no-pager`` 
[logJournalctlAfterShutdown.txt](https://github.com/robotology/icub-tech-support/files/5282263/logJournalctlAfterShutdown.txt)

@mbrunettini do you have any hint about something else we could check?

julijenv, 2020-09-28:
It has been restored the broken cables

IITCODE | Qta | Description
-- | -- | --
4766.1 | 1 | elbow cable N9
5964 | 1 | head cable B
5965 | 1 | head cable A

from hat i know eerything is working fine since then

closed 2020-09-28 by julijenv