robotology/mesh-support #991: icubosaka01: icub blinking does work as expected.

YANGGANG1971, 2020-09-25:
Following this walkthrough [guide](https://github.com/robotology/QA/issues/429), at the beginning, the icub could blink normally.  But after half an hour, the error occurred and see blow. The eyelid cannot open enough after finishing blinking. Is there something wrong with hardware? 
[image]

pattacini, 2020-09-25:
Actually, it's very likely so @YANGGANG1971.
Thus, I've moved the issue into our mech ticketing system.

kt10aan, 2020-09-29:
iCubOsaka01 has done a LOT of blinking in its life :). Sometimes, after extended blinking sessions the following happens:
- The grey cable inside the head is moving around enough (especially if the blinking is combined with sudden head movements) and it starts touching the small arm of the servo on the side of the head
- As that sole arm is responsible for the blinking of both eyes, it starts moving under increased friction from the cable. At some point, the behavior shown in the above photo starts to happen.
- Eventually, the arm will slide off its slot on the side of the head, further impairing the blinking action.

Fix: Open the head, realign the small arm, move the cable in a place that is not going to impeded the moving of the arm. Turn on the robot and observe the blinking action from the from and the back. When satisfied wit the result (i.e., that the arm works without any obstructions(, carefully put the head cover on again.

julijenv, 2020-09-29:
Dear @YANGGANG1971 ,
as suggest by @kt10aan , please remove the back cover of the head and the face cover. Investigate on why the movement is blocked by launching the blinking face off.
I will be available next monday on skype if you cant figure out what is the root of the problem.
let us know

YANGGANG1971, 2020-09-30:
@julijenv @kt10aan Thank you. From your answers, now I understood the problem was that the cable impeded the moving of the arm. The arm controlled the blinking. So.. ? I need to open face and head cover. In the process of opening face and head, What kinda any special things I need to pay more attention to in case of preventing destroying iCub?

YANGGANG1971, 2020-10-05:
Today, following your suggestion, we checked the iCub. We found although with no obstruction, we kept face cover away from the head, the blinking stuck is still here. See below for more info. Because every 10 s, we could hear one sound that is produced when iCub was trying to blink. Thus, the blinking controller could be fine. 
So we inferred that the problem should be caused by the motor this control iCub blinking. 
https://drive.google.com/file/d/199H5rR8becxfsnimHAxzI6J-Y-Nknm47/view?usp=sharing

Could we know the type of motor to replace it? Or how do we fix it in a better way?

julijenv, 2020-10-05:
Hi @YANGGANG1971 ,
FRom the video it looks like the eyelids start with the eyes shut. and when some commands are launched it gets a bit opened. So I dont think the motor is dead.
It is probably due to some data erasment form the flash memory of the eyelids controlling board. (it is something that has already been seen in various robots in the past).
Or as the sound from the video is quite bad - or at least the sound coming from the motor, maybe the motor is not in a good state.

YANGGANG1971, 2020-10-08:
[image]

YANGGANG1971, 2020-10-08:
[image]

YANGGANG1971, 2020-10-14:
Thank @julijenv. Currently, Alexis and I fixed the broken part. 
But it seems the single blinking does not work well as follows sometimes. We used the command line to control blinking. It should be one single blinking. From the video,  you can see that the iCub wants to close and but it cannot sometimes. So it seems double blinking. In fact, it is just a single blinking. Is there anything we do not notice to cause this? like not perfect gear coordination between the plastic arm and micro servo gear? Or it is normal and we care too much? 
https://drive.google.com/file/d/1yauodxEWQKA9Omaaq1WimE_JzJtSgIJg/view?usp=sharing

julijenv, 2020-10-16:
Hi @YANGGANG1971 and Alexis (he may need to specify is name inside the github even if his pesudo is strange because or else there is noway you can find him straight away),
I dont recall having seen so much blinking - Are susing the function blinking or are you sending blink every specific time period the command?)
As long as the blinking is working I would not care that much, but try to remove the cover adn send just one blink at a time every 5 sec just to see if the double blink is here or not. let me know

pattacini, 2020-10-16:
cc @Alexis00

Alexis00, 2020-10-19:
Ciao @pattacini , @julijenv !! Thank you for your comments! 
We will double-check the way we are sending the commands, but the software hasn't changed from before the hardware modifications. Therefore, we think it might be either a gear-teeth matching problem (after replacing the broken plastic part) or, with a lower chance, a low level software problem (i.e. in the blink controller). 
We will check and get back to you. Grazie mille~

Alexis00, 2020-10-19:
FYI, here are pictures of the replaced part. I have slightly redesigned the part, making it thicker in some places, in the hope to prevent future breaks. And I have 3D-printed several parts because they tend to break during the aluminum pins installation. 

[image]
[image]
[image]

YANGGANG1971, 2020-10-19:
> I dont recall having seen so much blinking - Are susing the function blinking or are you sending blink every specific time period the command?)

In fact, it is not controlled by every specific period. It is was controlled manually using[ iCubblinker ](https://github.com/robotology/funny-things). The sequence of command is as follows. 

> yarp write /write /iCubBlinker/rpc
> stop
> blink
> blink

Each "blink" command will make one real single blinking. 

>As long as the blinking is working I would not care that much, but try to remove the cover adn send just one blink at a time every 5 sec just to see if the double blink is here or not. let me know.

In fact, we tried setting the iCubblinker as "--autoStart" mode and blink every 10 sec.  
http://robotology.github.io/funny-things/module/iCubBlinker.html
Normally, one of 5 blinks is always a double blink. But we saw more double blinkings that one of 5 blinks. It is hard to verify the problem, so we only sent one command to test. 
Note: this trial did not remove the cover before.

julijenv, 2020-10-20:
Hi guys, (@YANGGANG1971 , @Alexis00,
I dontreally use the iCubBlinker when tested the face expressions I just use the .cycle.sh script that you could launch from icub-main/app/faceExpression/scripts and this just close the eyes once evry 10 sec and I vever see the double blink with it. could you pleas try to run it as well and let me know if you jsut get one blink.
let me know

YANGGANG1971, 2020-10-20:
Of course. We will test following your suggestions.
But I think iCubBlinker should work well. Because before the plastic part did not break, the single blinking works fine using iCubBlinker to control.

julijenv, 2020-10-20:
> Of course. We will test following your suggestions.
> But I think iCubBlinker should work well. Because before the plastic part did not break, the single blinking works fine using iCubBlinker to control.

So I assume though that something has not been properly replaced (or at least just a little tweaking of the opening and closing value of the servo motor could prenvet this)
please sue the software already metionned in first reply to change the extremity of he opening and closing and retry, and let m enkow

YANGGANG1971, 2020-10-21:
Following your suggestions, I tried using ./.cycle.sh script to control the blinking, but it seems this only controls facial expression by looping over each possible facial expression. See the video below for more info. 
https://drive.google.com/file/d/1XCuwRXAA_b0s_KWPjy7ZBEIjhUeq_ZuO/view?usp=sharing

I checked the code in  ./.cycle.sh. 
It did control facial expression. 

```
#!/bin/bash

list="hap evi shy cun sur ang"

while true; do
	for a in $list; do
		echo "set all $a"
		echo "set all $a"  | yarp rpc /icub/face/emotions/in
		sleep 2
	done
done
```

YANGGANG1971, 2020-10-22:
But it is very strange that when I tried using iCubBlinker to control Blinking, it seems working well again and no double blinking mixed. 
For more details, see the video below.
https://drive.google.com/file/d/1PwZ-7hpBmLKBsaCRrHolfz93PsECcFzZ/view?usp=sharing

YANGGANG1971, 2020-10-22:
The last link and files are not correct and please discard that! Sorry about that! 
Please check this link below.
https://drive.google.com/file/d/1WawawXnK9fmHi06KFQJCvdj-5IyiCZRs/view?usp=sharing

julijenv, 2020-10-22:
Dear @YANGGANG1971, 
from the video you sent me, it seems that the blinking is working as excepted.
As explained in my last psot the .cycle.sh should close the eyelids. becasue the command "hap evi shy cun sur and ang" or metacommand that send command to both the facial expressions leds and the servo motor.
If the blinking is working fine I guess that we could close the issue though?

YANGGANG1971, 2020-10-23:
Yes, it seems blinking working well. Thanks a lot. 
But it is weird to me that it seems using the .cycle.sh cannot close the eyelids or blinking as follows. 
https://drive.google.com/file/d/1XCuwRXAA_b0s_KWPjy7ZBEIjhUeq_ZuO/view?usp=sharing
If it is correct, on my side, I have no question about the blinking issue and we can close the issue. Appreciate it.

julijenv, 2020-10-23:
Hi @YANGGANG1971 , could you please try to open the emotions.ini files and see what is written inside.
this file is inside the same path as cyle.sh. from that path on a terminal just type
```
cd ../conf/
ls
cat emotions.ini
```
let me know if the command close eyes is present inside.

YANGGANG1971, 2020-10-26:
For details, see below. 

```

//number of facial expressions
emotions 10

//configuration of facial expressions
//index name Left-Eyebrow(HEX) Right-Eyebrow(HEX) Mouth(HEX) Eyelids(% open)
//index starts at 1

E1   neu 02h 02h 08h 38h
E2   ta1 **h **h 08h **h
E3   ta2 **h **h 16h **h
E4   hap 02h 02h 0Bh 70h
E5   sad 02h 02h 38h 38h
E6   sur 08h 08h 16h 38h
E7   ang 01h 01h 38h 70h
E8   evi 01h 01h 0Bh 70h
E9   shy 04h 04h 38h 70h
E10  cun 04h 01h 38h 38h

```

julijenv, 2020-10-27:
Hi @YANGGANG1971 ,
as you can see from the file you edited on you last reply, the eyelids (%open) changeed in between emotions so it meanas that the eyelids should nowmally change states in between the different facail emotions.
let me know your thoughts

YANGGANG1971, 2020-10-28:
Based on what you said, it is really strange. Although %open seems to be added to the program, the icub did not show eyelid open and close. 
https://drive.google.com/file/d/1XCuwRXAA_b0s_KWPjy7ZBEIjhUeq_ZuO/view?usp=sharing

violadelbono, 2021-03-17:
Hi @YANGGANG1971, what is the status of this issue? Can we close it? Thanks

Alexis00, 2021-03-18:
Hello @violadelbono ! Thanks for reaching out. Yes, go ahead and close the issue please. Take care and thanks again.

closed 2021-03-18 by violadelbono