robotology/mesh-support #993: Robotics X - Icub's thumb proximal cables are broken

tKsome, 2020-09-30:
## Brief description of the request/failure
Just like this issue https://github.com/robotology/icub-tech-support/issues/979, we have check the thumb proximal cables, it had broken, probably. Therefore, we decided to reopen a new issue and solve this problem.

the thumb cables of both hands are probably broken. in https://icub-tech-iit.github.io/documentation/icub_tendons/remove/ have the arm disassembly tutorial. BUT it is so complicated that we don't want take it apart with ourselves. We need guidance.

## Accompanying material
[image]

julijenv, 2020-10-01:
Hi @tKsome ,

could you please wait for me, from Monday I' will be totally dedicated to you guys
cheers,

tKsome, 2020-10-10:
Thank you for your help. Cable has been reinstalled, but there is a new problem. The reinstalled forearm will shake. joint3 in yarpmotorgui.

[1602337511122.zip](https://github.com/robotology/icub-tech-support/files/5359104/1602337511122.zip)

julijenv, 2020-10-12:
Hi @tKsome ,
I was not present on your skype meeting on Friday but from the video there 2 possibilities either the cables of the elbow are loose or, from what happened often, the 2 lateral pieces holding the forearm stand are loose.
Could you check if the cables tensioners are tightened and if those 4 following screws are tightened as well?
[image]
[image]
let us know

violadelbono, 2020-10-12:
I also suggest to check that the three screws holding the elbow are tight (I remember you had some issue while removing the forearm and pushing hard on those threads)

[image]

julijenv, 2020-10-15:
@tKsome , Did you succeed to fix the shakering of the elbow by checking the tensionning of the s crews above mentionned?
thx to let us know

tKsome, 2020-10-20:
awwww, we haven't test it yet. but we will test it later according to your suggestions

closed 2021-02-18 by violadelbono