robotology/mesh-support #994: iCubLyon01 - Dijon - replacing CAN bus filter in right arm - need advice

pfdominey, 2020-09-30:
## Brief description of the request/failure
Julien helped me to replace cable 7, but then we discovered that a wire was broken in the right arm, in a CAN bus filter.  I have recieved the part, and will install it, but need some help.

## Detailed context
On the CANbus filter on one side there are two places where the cable can be inserted.  Under the transparent plastic there is on one side a black electronic part, and on the other side there are the wires that are more visible.  Looking at this picture, it seems that the cable should connect on the side where there is the small black part (the upper side on the attached figure, pointed by the blue arrow) and not on the side with the wires and not the small black part, but I am not sure.

can you confirm that the N2 wire should go into the side that has the square black electronic component under the clear plastic?

Thanks

[image]

## Accompanying material

julijenv, 2020-10-01:
Dear Prof. Dominey,
I would be reachable on Skype from Monday not prior.
please reach me on morning if possible.
BTW from the picture, the connections with the boards look alright.
C U on Monday

pfdominey, 2020-10-01:
Perfect - thanks Julien. Probably we will come to Dijon robot lab on monday and arrive end of the morning.  So whatever is convenient for you, monday afternoon or tuesday morning.  Let me know if that is ok.
Thanks - cheers

julijenv, 2020-10-15:
Dear Professr @pfdominey ,
could youi please try to take the same picture that you have made for the issue creation in order to make me check the connection and also please try to make another focus on the board and new dfilter.
thx inadvacce

Uboldi80, 2021-10-05:
can line fixed.

see the issue:  https://github.com/icub-tech-iit/proto/issues/1318

closed 2021-10-05 by Uboldi80