robotology/mesh-support #995: Support request on starting the iCub3 robot including the legs

S-Dafarra, 2020-09-30:
## Brief description of the request/failure
We would need some support in starting the iCub3 robot we have in the lab. In particular, I am not sure about:
- the current software status
- where are the parameters stored
- how to switch it on
- if the legs are working

CCing interested users and people that know something
@DanielePucci @GiulioRomualdi @isorrentino @prashanthr05 @xEnVrE @vtikha @pattacini @ale-git

pattacini, 2020-09-30:
Given the nature of your request @S-Dafarra , would you be so kind to open this issue in https://github.com/icub-tech-iit/task-force-miscellanea instead?

Unfortunately, transferring an issue from one org to another is not (yet) allowed by GitHub. Therefore, just do copy'n'paste. I'll leave this to you so that it'll remain recorded that you're the author.

S-Dafarra, 2020-10-01:
> Given the nature of your request @S-Dafarra , would you be so kind to open this issue in https://github.com/icub-tech-iit/task-force-miscellanea instead? I've put this private repo into our automation for dealing w/ outside collaborators.
> 
> Unfortunately, transferring an issue from one org to another is not (yet) allowed by GitHub. Therefore, just do copy'n'paste. I'll leave this to you so that it'll remain recorded that you're the author.

Sure, I opened https://github.com/icub-tech-iit/task-force-miscellanea/issues/50. Closing this then.

closed 2020-10-01 by S-Dafarra