robotology/mesh-support #996: icubosaka01: iCub lelf leg is broken.

YANGGANG1971, 2020-10-05:
See the figure below. The tendon of iCub left leg is broken. What should we buy to fix it? Or How do we fix it? 

[image]
[image]

julijenv, 2020-10-05:
Hi @YANGGANG1971 ,
could you please contact me on skype, my ID is icub.support.
let's talk about your 2 problems
#991

YANGGANG1971, 2020-10-05:
Appreciate it.  Let's discuss it on Skype.

julijenv, 2020-10-05:
Hi @YANGGANG1971 , please follow the instruction of our official HOWTO restor iCub [website](https://icub-tech-iit.github.io/documentation/icub_tendons/waist_legs/)

Alexis00, 2020-10-05:
Hello @julijenv ! Long time no talk! If you guys haven't talked yet, maybe we can try to meet tomorrow or the day after. Let us know your schedule

julijenv, 2020-10-05:
Hi @Alexis00 ,
with @YANGGANG1971 , we discussed over skype and we agreed to get in touch on Thursday 9:30 am CET for the face cover #991 , for the elgs instead I think, please refer to my last message to fix the ankle or if you want to still work with Cub in the meanwhile just disable or remove every line mentioning the legs on the icub-all.xml startup configuration file.
let me know

julijenv, 2020-10-15:
Hi @Alexis00 , @YANGGANG1971 ,
- did you succeed to replace the plastic lever of the eyelids?
- did you check if you have the cable needed to repair the left leg?
let us know

YANGGANG1971, 2020-10-15:
Hi @julijenv, Alexis has replaced the plastic lever of the eyelids. But it seems there is a minor problem and see below.
https://github.com/robotology/icub-tech-support/issues/991

It seems we do not have the cable for the leg and we only have cables for arms. Maybe @Alexis00 could explain better than me.

Alexis00, 2020-10-15:
Hello @julijenv ! Thanks for checking with us! 
Indeed, as @YANGGANG1971 said, we do not seem to have any replacement cables for the legs (because the lab generally doesn't use the legs for experiments, and maybe that is why the tendon broke). 
Therefore, we would need to replace that tendon. By checking on the link that you so kindly provided, I think we would need either Cable 15 or 16. Is there a "Repair Set" that we need to buy to get those cables? Or do we need to request individual replacement parts? Thanks!

Alexis00, 2020-10-22:
Hello again @julijenv - This is a kind reminder about this item. When you have time, please advise us on getting new cables for the boot of the robot. Thanks!

julijenv, 2020-10-22:
Hi @ALexis00,
could you please make a list and the number of each of cable you still have as spare parts.
It is important to figure out if you are in need of spare parts for future repairs.
there are 2 solutions actually to get spare from us,
 - if you have few cable remainings it would be better to get a set of tendons (from head to legs) - I will ask for the administration office to give you a quotation for your robot - **this set of cable is not customizable**.
 - if you just have this cable missing we may find a temporary solution.
let me know

Alexis00, 2020-10-29:
Hello again @julijenv ! Sorry for the late reply. 
We do have several tendons, but mostly for the upper body. I think it would be good for us to buy a complete set (if it falls within budget for the laboratory). It would be better for us to have spare tendons.

So please, let me know when you have the Quotation for the set for our robot. Thanks!

Alexis00, 2020-11-25:
Hi @julijenv ! Any update regarding the Quotation for the set of tendons for our iCub (not customizable)?

Thank you in advance! Take care

julijenv, 2020-11-25:
Hi @Alexis00 , I already sent a mail to the purchasing office, I'm going to check if it had been done,
I put Yukie as the main receiver of the quotation in my mail. Did she by any chance receive anything from us (IIT) about it?
best regards.

Alexis00, 2020-11-26:
Hi @julijenv , I've asked Yukie but she tells me she hasn't received any quotation yet. Please check with the purchasing office when you have the chance. Thanks again for all your help!

violadelbono, 2021-05-18:
Cables shipped. Closing

closed 2021-05-18 by violadelbono