robotology/mesh-support #1000: [iCubGenova04] The robot fault itself from time to time

S-Dafarra, 2020-10-06:
## Brief description of the request/failure
While using the robot, sometimes it gets faulted even if no one touched the red button. This happened both while walking, but also while on air.

## Detailed context
When it happens, the status LEDs remain green as normal, but we noticed that the ``yarprobtinterface`` was detecting the external fault pressed. 

This may be related to #990 

## Accompanying material

Unfortunately, we did not manage to save any. We were running it from the terminal and the output was too bug due to #997 

cc @DanielePucci @maggia80 @MrAndrea @GiulioRomualdi @prashanthr05

GiulioRomualdi, 2020-10-06:
Issue number 1000! Congrats!

DanielePucci, 2020-10-06:
@S-Dafarra @GiulioRomualdi we should check if it is a problem of the LAN connection a possibly use the cable @gabrielenava was mentioning. Also, a possibility could be to go WiFi

julijenv, 2020-10-19:
Hi guys, (@S-Dafarra , @DanielePucci , @GiulioRomualdi )
we changed this morning the ethernet plug, if it is ethernet related, I guess that with this new plug the odds are that it will occur less.
keep us informed.
It is coming from the Jack of the fault button, as  @S-Dafarra , suggested maybe it was related to the fall from #990 . I already check the jack itaself and it looks like OK to me. SO most probbly it is the controlling board.

DanielePucci, 2020-10-19:
Thanks @julijenv

lrapetti, 2020-10-20:
It happened again today once, here is the log: [log_termination.txt](https://github.com/robotology/icub-tech-support/files/5409959/log_termination.txt).

I noticed it happen as I touched the backpack (I was opening it to check the battery), I don't know if it is a coincidence or if there is any unstable contact,

julijenv, 2020-10-21:
It has been checked thoroughly and it looks like the jack of the fault button if not plugged correctly could cause this behaviour - I just want to warn you that logically the fault cable should also be supported by a tie-wrap because unfortunately the weight itself from this cable if pulled just a little bit, could make it almost unplugged.
It is the utmost importance to check the jack well-plugged every time you launch the robot and make sure the cable is firmly retained by a tie-wrap at the level of the power cable to prevent from unwanted removal.
@lrapetti @GiulioRomualdi  @S-Dafarra @gabrielenava

S-Dafarra, 2020-10-21:
> It has been checked thoroughly and it looks like the jack of the fault button if not plugged correctly could cause this behaviour - I just want to warn you that logically the fault cable should also [...]

To be honest I am a bit suspicious about this. The jack has always been the same, but it is the first time we experience this. In some cases, it happened also when the robot was barely touched. If you think that the connector is the cause, maybe the connector is faulty or damaged.

lrapetti, 2020-10-29:
The problem seems to be still there as we experienced it at least a couple of time per day. I think it needs to be investigated further, and, indeed, following the comment from @S-Dafarra I am reopening this issue.

We may add here some further observations that can eventually help in debugging the problem:
- this happened also while the robot is simply hanging and nobody touches it.
- we have been working all the time with the battery, and I think the problem is happening mainly when the battery is attached to the charger.

Please @kouroshD @gabrielenava @CarlottaSartore let us know if you observe again this failure even when the robot has power cable (not battery).

julijenv, 2020-11-02:
Hi @lrapetti , @S-Dafarra  and co.,
please add log everytime this happen to be able to narrow down on the real problem.
thx in advacne

julijenv, 2020-12-30:
Hi guys, could you please let me know if you experienced this issue when the robot is plugged with the cable or if it is only when using the battery pack as was suggested @lrapetti ?

S-Dafarra, 2021-01-04:
> Hi guys, could you please let me know if you experienced this issue when the robot is plugged with the cable or if it is only when using the battery pack as was suggested @lrapetti ?

If I remember well, this happened also with the cable, but @kouroshD @gabrielenava @CarlottaSartore please confirm.

gabrielenava, 2021-01-04:
>If I remember well, this happened also with the cable, but @kouroshD @gabrielenava @CarlottaSartore please confirm.

Not sure, but we observed possibly related issues like https://github.com/robotology/icub-tech-support/issues/1046 when the robot was with the cable.

lrapetti, 2021-01-04:
> Not sure, but we observed possibly related issues like #1046 when the robot was with the cable.

This one looks a different issue. If you never experienced this failure, since we were experiencing it a couple of time per day, I would say it is most likely related to battery usage only.

closed 2021-06-23 by github-actions[bot]