robotology/mesh-support #997: Left fingers are not calibrated

GiulioRomualdi, 2020-10-06:
## Brief description of the request/failure
When I started the robot the left fingers are not calibrated 

## Detailed context
Opening the `iCubGui` the fingers position are the following
[image]

The first error thrown by the `robotinterface` is:
```
16755 42.127201 <ERROR>  from BOARD 10.0.1.25 (left_arm-eb25-j8_11), src LOCAL, adr 0, time 125s 840m 387u: (code 0x02000010, par16 0x0001 par64 0x0000000000000000) -> MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
 
16757 42.127310 <INFO> created device <embObjMotionControl>. See C++ class yarp::dev::embObjMotionControl for documentation.
 
16758 42.127324 <ERROR>  from BOARD 10.0.1.25 (left_arm-eb25-j8_11), src CAN1, adr 0, time 125s 842m 109u: (code 0x00000019, par16 0x0301 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .
 
16762 42.127378 <ERROR>  from BOARD 10.0.1.25 (left_arm-eb25-j8_11), src CAN1, adr 0, time 125s 843m 108u: (code 0x00000019, par16 0x0301 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .
 ```
## Accompanying material

[log_icub-head_yarprobotinterface_8979.txt](https://github.com/robotology/icub-tech-support/files/5333046/log_icub-head_yarprobotinterface_8979.txt)

cc @S-Dafarra and @prashanthr05

S-Dafarra, 2020-10-06:
Cc @violadelbono

violadelbono, 2020-10-06:
Hi @GiulioRomualdi and @S-Dafarra 

Unfortunately I am not in the lab today. We will check as soon as we can return.
I don't know if this can be due to a misconnection of the board 25 when attaching the arms yesterday.

julijenv, 2020-10-07:
Hi guys,
from the logfile it looks like quite easy to guess that the MAIS is not talking. either the CAN line is broken or the MAIS is not powered.
@GiulioRomualdi @S-Dafarra

julijenv, 2020-10-19:
Hi guys, (@GiulioRomualdi  ad @S-Dafarra )
could you please check that it works now.  By solving the CAN bus line issue( #1008 ) I am convinced that the fingers shoud be alright now.
let us know and feel free to close if ever it is fixed

kouroshD, 2020-10-20:
Hi @julijenv 
Thank you for fixing the left fingers. I was testing the robot this morning, there is the `l_thumb_distal` which does not configure, and the `l_index_proximal` is in hardware fault.

julijenv, 2020-10-20:
Hi @kouroshD ,
I will give an eye today after the meeting.

lrapetti, 2020-10-20:
This afternoon I tested the robot (after https://github.com/robotology/icub-tech-support/issues/1012 happened), and I noticed the left hand was not started correctly, the following problems were observed:
- `l_finger` it started but I cannot move it with yarpmotorgui
- `l_thumb_distal`, `l_index_distal`, `l_middle_distal` start in idle (after switching to position they were fine)
- `l_little_finger` is not_configured

Here is the log: [log_fingers.txt](https://github.com/robotology/icub-tech-support/files/5410195/log_fingers.txt)

julijenv, 2020-10-21:
Hi @lrapetti  @kouroshD 

> * `l_finger` it started but I cannot move it with yarpmotorgui

**the motor is a little bit old and from what I know it has never been really completely tweaked the PID and PWM to make it work the best it could - it is an already existing internal issue adressed to team-fix. The overall situation due to a noisy Hall sensor at the level of the pinkie finger is as when the value goes out of boundaries the joint goes in HW FAULT please consider reput the fingers in the boundaries to have the rest of the fingers calibrating at start**

> * `l_thumb_distal`, `l_index_distal`, `l_middle_distal` start in idle (after switching to position they were fine)

**Unfortunately because the joint 7, 9, 11, and 13 are coupled at the level of the calibration if ever one of them is not calibrating it prevents the rest of them to be turned ON and put them in IDLE or not_congfrgured**

> * `l_little_finger` is not_configured

**THis joint is also a bit tricky, often the mechanical part (base of the pinkie finger) tends to wear out and the calibration gets off.
Either you need to recalibrate the joint or just close it a little bit in order to put it back inside its range of function and click on idle and run and move it until it close - it is now reabilitated, you can now play with it** 

Hope this help

julijenv, 2020-12-30:
I think that the #1040 has solved this issue.
feel free to repoenend if not satisfied @lrapetti

closed 2020-12-30 by julijenv