robotology/mesh-support #1001: iCubLyon01 robot won't start - firmware version error

pfdominey, 2020-10-07:
## Brief description of the request/failure
Running yarprobotinterface fails.  Logger shows red errors related to firmware
[yarprunlog_07_10_2020_14_03_13.log](https://github.com/robotology/icub-tech-support/files/5340501/yarprunlog_07_10_2020_14_03_13.log)

## Detailed context
[yarprunlog_07_10_2020_14_03_13.log](https://github.com/robotology/icub-tech-support/files/5340505/yarprunlog_07_10_2020_14_03_13.log)

## Accompanying material

julijenv, 2020-10-07:
Dear prof. @pfdominey ,
checking the log file it seems that one more time as for #994, it seems that the first MC4 board is not responding.
I suggest you to reopen the upper arm cover - maybe even removed it and check the cabling for the specific board already pointing out by the last issue.
let us know

pfdominey, 2020-10-08:
Dear Julien,
We took the arm cover off, and inspected the replaced CAN filter, and other wires.  We could not see anything wrong.
Do you think there is any possibility of sending someone here to fix the robot?
Thanks
Peter

julijenv, 2020-10-08:
DEar Professor @pfdominey ,
could you please try to take a bunch of photots related to connector going into the CAN filter, the connector of the CANfilter itself before and when plug into the MC4 board (the board controlling the motors). and a last one of the power connector of the MC4 board (raindow colored one - 8pins). waiting for yor pictures.