robotology/mesh-support #1002: [iCubChemnitz01] Network Connection fails

torsten93, 2020-10-07:
## Brief description of the request/failure
After a longer time a tried to use the robot today, there a problem occured. 
I could start the CPU of the robot and it seem to boot and first I got a network connection, but when I want to start yarprun on pc104 it failed to start. 
Then I tried out to remove the two last lines in hosts and to start it, it does not worked. Even worse, since the, when I start the icubsrv Laptop and the robot I didn't get a network connection.

## Detailed context
We use the setup shown in the image below with a switch connecting the robot and the Laptop in a local network.

What could be possible settings, which need to be changed?

## Accompanying material
[image]
[image]
[image]
[image]
[image]

julijenv, 2020-10-08:
Hi @torsten93 ,
I am not really the network master but I am going to ask around before giving you a finla answer.
in the meanwhile could you try to set up first your switch to be 10.0.0.252. the same network as the icub-head or icubsrv. and just ty to relaucn the head and see.

mbrunettini, 2020-10-08:
the wired connection that connects the icubsrv to the pc104 MUST be configured with static IP with the following parameters:
IP ADDR : 10.0.0.1
Netmask : 255.255.255.0
See this [link](https://wiki.icub.org/wiki/ICub_server_laptop_installation_instructions)

torsten93, 2020-10-12:
Thank you for the responses. Since I am in the university to work on the iCub at Wednesday, I will try and report then.

torsten93, 2020-10-14:
We tried to run the iCub today:
-> the config helped to fix the network issues on the icubsrv-laptop
-> we couldn't reach the iCub pc104 further and tested the USB drive plugged in the iCub head at the laptop and it seam to be damaged -> we will try to fix this, for this we may need to download the linux image, but the link (http://wiki.icub.org/iCub/downloads/pc104-images/) given in the respective documentation (http://wiki.icub.org/wiki/The_Linux_on_the_pc104) does not work (I only get a "Page not found")

julijenv, 2020-10-15:
Hi @torsten93 ,
I, unfortunately, have the sad news to announce you, the pc104 images have an issue lately - they dont get installed - I got the news from my colleague. In the next days I will be able to give you more insights on how long it will take to fix them.
In the meanwhile, try to see if you are able to rstored the one you have, probably just hte grub or something of this sort is damaged.
I will keep you informed.
cheers,

cc @mbrunettini

mbrunettini, 2020-10-15:
Just to check what's wrong in the bootable USB :
1. open the iCub Head
2. take out the USB memory which contains the OS
3. plug the USB in any laptop
4. boot from the USB

So you can check what's the problem.

@julijenv can assist you in steps 1 & 2, I can help in the rest of the process

torsten93, 2020-10-21:
Thank you, we replaced the USB with a new one and put a backup on it. Now it works correctly.

closed 2020-10-21 by julijenv