robotology/mesh-support #1004: iCubGenova09 - request for general maintenance

xEnVrE, 2020-10-08:
## Brief description of the request/failure

This morning the robot iCubGenova09 has been tested after many months of inactivity. We report here some HW issues:
- some optical encoders are dirty
- some optical encoders have broken phases
- some absolute encoders do not respond

I am attaching a detailed log from the `yarprobotinterface`.

## Accompanying material

[yarprunlog_08_10_2020_11_02_02.log](https://github.com/robotology/icub-tech-support/files/5346560/yarprunlog_08_10_2020_11_02_02.log)

cc @pattacini, @S-Dafarra, @ale-git, @isorrentino, @maggia80 , @DanielePucci

S-Dafarra, 2020-10-13:
Hi @julijenv @violadelbono do you have any estimate/plan for this activity?

julijenv, 2020-10-19:
Dear All, (@S-Dafarra , @xEnVrE , @pattacini  and @ale-git )
After having unmounted all the cover to check every single sensor attached to the 2FOC line we are quite convinced that the problem with the robot when it does not start some joints is due to a FIRMWARE PROBLEM.
Here is the list of sensors on which for about 10 differents calibration with yarprobtinterface where put in HW FAULT for the problem quote below:

> MC: 2FOC quadrature encoder phase broken. The motor encoder is not counting even if the motor is moving, please check motor encoder cables.

list of complaints for sensors in HW FAULT:

|body part|joints|
|---|---|
|left_arm|j1,j2,j3|
|right_arm|j1|
|left_leg|j0,j1,j2,j3,j5|
|right_leg|j0,j1,j5|
|torso|j0,j1|

Just once it has been found that there was probably a reading error due to dirty disk. (this can be found by trying to reabilitate the joint and make it moved with yarpmotorgui BUT if it turned HW FAULT again it means that it is dirty disk (the error message showed up just in this case) BUT IF IT MOVES CORRECTLY IT MEANS THAT THE 2FOC READINGS AT START WERE WRONGLY DONE!!!   -> perhaps make more readings at start could solve this behaviour).

In any case, from our point of view the robot is working, the only little problem is that if the robot needs to retrieve the control from the joint that was put in HW FAULT during calibration, it needs to be reablilitated thanks to yarpmotorgui (IDLE + RUN), the trade-off is unfortunately that it is quite unconveniente to see the robot not calibrated by itself and I believe that the PWM are lower down when the calibration of the joint dont go well.

let us know

maggia80, 2020-10-19:
@julijenv the calibration of iCub3 is a bit different from iCub2.5. The brushless motors don't have the HALL SENSOR. There is a reflective encoder in the rotor with a Index in a specific position. At startup, it's applied a specific voltage pattern to the windings of the stator in order to move the rotor till the index is found. Once the index is detected, the motor can be controlled by means of the reflective encoder. The message:

`MC: 2FOC quadrature encoder phase broken. The motor encoder is not counting even if the motor is moving, please check motor encoder cables.  ` 

is not clear to me. How is the motion of the motor detected? By reading the absolute encoder in the joint?  

@ale-git , @marcoaccame could you help us with this?

thanks.
Marco

ale-git, 2020-10-20:
Hi @maggia80, the motor performs a full rotation, detected by the Index, with no readings at all from the quadrature signals. In this way the fw can detect a quadrature phase fault. In the same way, if the ticks are much less than expected, the "encoder dirty" fault is detected.

maggia80, 2020-10-20:
> Hi @maggia80, this message is sent when the motor performs a full rotation, detected by the Index, with no readings at all from the quadrature signals. In this way the fw can detect a quadrature phase fault. In the same way, if the ticks are much less than expected, the "encoder dirty" fault is detected.

ok I see. it's something like that? :

[image]

In this case it sounds strange that there are encoder phase error from time to time. Could it be because the rotor is aby chance already on the Index?

maggia80, 2020-10-20:
[Uploading encoder_signals.zip…]()

ale-git, 2020-10-20:
Yes the scheme is correct. About the last question, I'm going to check the fw sources, but I suspect that it can't happen unless there is a bug.

ale-git, 2020-10-21:
@julijenv the firmware is not changed since the robot was performing yoga-like movements at fast speed and the IOL demo as well, thus at least there must be some interaction with hardware problems since a firmware problem can't emerge from nowhere...

isorrentino, 2020-11-11:
Hi @julijenv, the last few times we used the robot only one motor has never correctly started. The motor is the `right ankle roll`. The error message is 
```
MC: 2FOC quadrature encoder dirty. The number of thicks in a complete revolution of the motor was 
lower than expected, the optical disks need to be cleaned. In par64 0xFF is the mask of raw encoder value.
```

Even if I put it in idle and then again in position control it goes again in HW fault.

[Here the log.](https://github.com/robotology/icub-tech-support/files/5525338/log_icub-head_yarprobotinterface_5280_2020_11_11_dirtyencoder_r_ankle_roll.txt)

cc @S-Dafarra @ale-git @DanielePucci @pattacini @maggia80

julijenv, 2020-11-12:
Hi @isorrentino ,
could you please check again today, I passed this morning and clean the sensor and the optical disk.
let me knw if it solve the issue

S-Dafarra, 2020-11-17:
> Hi @isorrentino ,
> could you please check again today, I passed this morning and clean the sensor and the optical disk.
> let me knw if it solve the issue

Hi @julijenv, I checked today, but the problem still occurs. Now it happens also on the right ankle pitch. Here the log
[log_icub-head_yarprobotinterface_8428_right_ankle_issue.txt](https://github.com/robotology/icub-tech-support/files/5553094/log_icub-head_yarprobotinterface_8428_right_ankle_issue.txt)

julijenv, 2020-11-19:
The rbot finally calibrate correctly today so I close the issue, feel free to reopen it or create a new one later

closed 2020-11-19 by julijenv

S-Dafarra, 2020-11-19:
Thanks @julijenv and all!