robotology/mesh-support #1005: [iCubGenova04] - point of attack of the chest cover to the shoulder is broken

gabrielenava, 2020-10-08:
## Brief description of the request/failure

The point at which the chest cover is attached to the shoulder got broken, and now the cover is shaky. I think it needs to be glued/replaced.

## Detailed context

The cover got broken during an experiment with velocity control.

## Accompanying material

julijenv, 2020-10-08:
Hi all,
@gabrielenava , @S-Dafarra , unfortunately, falling means breaking . the torso cover stand/bumper has been repaired at lesat 110 times ad replace 3 times..
It may be time for you to ask for creating a specific bumper just for you.
walking measn sometimes falling I know but we cant change everything everytime, This piece needs to be redisegnes to absorb the fall or else it measn that we will need to replace it to often.
cc @fiorisi

S-Dafarra, 2020-10-08:
> Hi all,
> @gabrielenava , @S-Dafarra , unfortunately, falling means breaking . the torso cover stand/bumper has been repaired at lesat 110 times ad replace 3 times..
> It may be time for you to ask for [...]

Notice that the front cover did not break because of the fall. We broke the back cover because of the fall. 

Regarding the fact that chest cover often gets broken, I opened an issue back in time https://github.com/icub-tech-iit/tickets/issues/136

gabrielenava, 2020-10-08:
Hi @julijenv, actually, on iRonCub there is @vpunithreddy which is working on a new version of the chest fixtures, maybe this will help in the near future

CarlottaSartore, 2020-10-16:
During today experiments, it get broken again 
C.C. @gabrielenava

julijenv, 2020-10-19:
The back cover is broken - the part attaching the front cover to the chest cover got broken, what should we do??? @DanielePucci

julijenv, 2020-10-19:
As discussed with @S-Dafarra , we decided to replaced the T shape of the chest as the back cover is broken and not repairable (or at least if you order a new one). We think by doing the change the chest cover will be hanging more safely but not secure as it was before, so be careful!!!
cc @violadelbono (to be filled up with more details, thx in adv)

violadelbono, 2020-10-19:
The T-shape component to be replaced is the following:

|  IITCODE |  ALIAS |REV |  DESCRIPTION | NOTES |
|---|---|---|---|---|
|  11406 | RC_IIT_010_P_082| A3  |  MCP chest assembly cover support| PVC|

[image]

[image]

S-Dafarra, 2020-10-19:
So to recap. The cover that is broken is the back cover (the one with the backpack). Substituting the back cover is expensive, especially in terms of time. An alternative is to substitute the piece shown in the comment above. In this way, the torso cover would be kept in place on one shoulder and by that T-shaped piece. The other shoulder would remain broken.

Would it be possible to substitute it while calibrating the FT sensors? @julijenv @violadelbono 

@DanielePucci @gabrielenava what do you think?

DanielePucci, 2020-10-19:
@S-Dafarra 
> @DanielePucci @gabrielenava what do you think?

Ok to me if also @gabrielenava agrees

gabrielenava, 2020-10-20:
Ok also for me, consider also that the T-shape is  one of the pieces that @vpunithreddy is redesigning for iRonCub, and that we are already about to produce a 3D printed version of the piece for testing purposes.

julijenv, 2020-10-20:
hi guys, (@GiulioRomualdi ,, @DanielePucci , @gabrielenava , @S-Dafarra ). To be able to proceed with replacing the T-shape meanwhile the robot will be stopped for recalibrating the FT sensors we need this piece that unfortinately we dont have in the warehouse, so I guess it is quite critical to try to ask for a favor from the MWS to make the piece tomorrow. or else it will done later.

DanielePucci, 2020-10-20:
Thanks @julijenv, let's talk about this!

vpunithreddy, 2020-10-20:
A [job](https://github.com/dic-iit/component_ironcub/issues/237) has been opened to produce the T-shape along with the fixtures designed for iRonCub, which might be able to solve the fracture issues being caused in the T-shape

julijenv, 2020-10-20:
The T-shape will be replaced in the meanwhile the FT sensor got recalibrated on Thursday

closed 2020-12-04 by julijenv