robotology/mesh-support #1006: [iCubGenova04] The yarprobotinterfaces crashes several times

GiulioRomualdi, 2020-10-09:
##  Brief description of the request/failure
We switched on the robot, we ran the `robotinterface` and then the `yarprobotinterfaces` segs fault.  We were able to store two logs (that it's really heavy because of #997)

## Accompanying material

[segfault_interface.zip](https://github.com/robotology/icub-tech-support/files/5356306/segfault_interface.zip)

@prashanthr05 and @S-Dafarra

julijenv, 2020-10-12:
Hi @GiulioRomualdi ,
Could you just let me know if for any reason you made update on Software and/or FW?

julijenv, 2020-10-12:
By reading at the log files, it is hard to understand why it stops (yes the CANline on the left arm is non connected correctly but I doubt it creates this crashes). Unfortunately you will need to investigate by your own the reason why the yarprobotinterface crashes, I mean you will need to launch one at a time the other softwar ecommunicating with the yarprobotinterface to understand which one block the yarprobotinterface.
What you are experiencing is due to SOFTWARE/FIRMWARE ISSUE and to resolve it I guess the 2 log files are useless are not appropriate.
@ugo, @marcoaccame , what kind of information would you propose them to give you to have any chance to narrow down the reason why the yarprobotinterface just stop this way.
thx in advacne to the overall peopole invoved

closed 2021-06-14 by github-actions[bot]