robotology/mesh-support #1007: iCubLyon01 left hand hardware errors

pfdominey, 2020-10-14:
## Brief description of the request/failure
Greetings, iCub is working, pointing with left hand, but the joints of the thumb and fingers signal harware error.

## Detailed context
recently we had to disable the right arm because of problems with the CAN electronics, probably related to a repair we made, replacing a CAN filter (thanks Julien!!)
I edited the startup xml file that species all the devices, to exclude right arm.  Pretty sure that I did nothing to left arm, but I include the xml file and the log file that shows the hardware errors.  Please take a look and let me know what you thinnk.  Thanks!
[yarprunlog_14_10_2020_18_15_32.log](https://github.com/robotology/icub-tech-support/files/5379637/yarprunlog_14_10_2020_18_15_32.log)

[icub_no_legs.txt](https://github.com/robotology/icub-tech-support/files/5379658/icub_no_legs.txt)
[image]

## Accompanying material

julijenv, 2020-10-15:
Dear Professor @pfdominey ,
could you please take a picture of yarpmotorgui to show me the fingers in hardware FAULT. I would recommend you also to send me the video of the calibration phase and try to focus on the left arm + hand.
thx in advacne

pfdominey, 2020-10-20:
[image]
[image]
[image]

Here is a link to the video
https://photos.app.goo.gl/mmqo6gSjD5i99AfH6

Hi Julien - here is the info.  Otherwise, the rest of the left arm is working great!  But no hand

julijenv, 2020-10-20:
JHi professor @pfdominey ,
the 8 last joints are RED meaning HW FAULT, i guess that the MASI has an issue could you double check that the Leds on the board above the knuckle of the left hand are enlighted?
if yes it means that the green or white electrical wire are not OK thugh.
let me know