robotology/mesh-support #1014: iCubLyon01 upgrade

pfdominey, 2020-10-21:
## Brief description of the request/failure
Our iCub has old pc104, old yarp (circa 2016) and is starting to experience issues.

## Detailed context
We previously discussed an upgrade, and you prepared an invoice.  I attach that here for reference.
Can we consider shipping iCub to IIT, and getting a general maintenence, upgrade.  This would also include discontinuing use of the iKart.  We would return to just a simple stand on wheels as we had in the beginning.
We could also send a new dell laptop that would serve as the console (or you could do that and charge us).
[Prot_0010956_18_Quotation_old_iCub_maint+upgrade_Dr._Peter_Ford_Dominey_INSERM_Lyon_(Offer_Rev_0) (1).pdf](https://github.com/robotology/icub-tech-support/files/5415952/Prot_0010956_18_Quotation_old_iCub_maint%2Bupgrade_Dr._Peter_Ford_Dominey_INSERM_Lyon_.Offer_Rev_0.1.pdf)

Please let me know your thoughts on all this.

## Accompanying material

julijenv, 2020-10-21:
#994, #1001, #1007

julijenv, 2020-10-21:
Dear @pfdominey ,
I will forward your mail to the person in charge to see if the conditions for this request are still OK - I am unfortunately not able to give you information at this regard.
I do believe though that it could be a better idea to send it here the robot instead of going to you to make this changes because  COVID and mainly spare parts.
I will let you know.

closed 2020-10-21 by julijenv