robotology/mesh-support #1015: iCubShenzhen01 left hand's thumb had a problem again.

tKsome, 2020-10-24:
## Brief description of the request/failure
After the repair of the broken cable, the thumb can already be controlled, but there are still cable stuck sounds. And When i want to give the thumb a wide angle, the motor keeps ringing(i think the motor is idling). And sometimes JOINT8 is no longer controlled after the maximum angle.  

## Accompanying material

we are afraid to do experiments now. if there is position control experiment later, i will send you a video via Skype

tKsome, 2020-10-26:
this problem should be the cable broken last time but we donnot find it. and it also broken in left hand
[1603708127211.zip](https://github.com/robotology/icub-tech-support/files/5438261/1603708127211.zip)

violadelbono, 2020-10-26:
Hi @tKsome 

From your pictures and video it seems to me that this time the thumb oppose cable broke (joint 8).

This motor is located inside the hand, so you will not need to remove the forearm (luckily!).
Here you can read the documentation related:

https://icub-tech-iit.github.io/documentation/icub_tendons/wrist_hand_cabling_V2/#21-location

I suggest you first ensure that the thumb oppose is the cable that is actually broken and then read the procedure carefully in order to repair it. We can call on Skype in case you need advice.

tKsome, 2020-10-26:
@violadelbono hi, we need you help, the broken cable cannot come off

violadelbono, 2020-10-26:
Call me on Skype

tKsome, 2020-10-27:
We fixed the tendon, but the left arm joint 8's encoder seems to be broken

julijenv, 2020-11-02:
Hi @tKsome ,
didi you get some good news from the sensor?
let us know

closed 2021-02-18 by violadelbono