robotology/mesh-support #1019: Problem with the hand motor feedbacks

kouroshD, 2020-10-29:
## Brief description of the request/failure
I have logged the encoder values and velocities of the robot hand, however it seems there is a sign mismatch between [encoder values](https://robotology.github.io/yarp-documentation/git-master/classyarp_1_1dev_1_1IEncoders.html#a8c5adefdc8c0627a7ecfe7f6d972deaa) and [encoder speeds](https://robotology.github.io/yarp-documentation/git-master/classyarp_1_1dev_1_1IEncoders.html#ac84ae2f65f4a93b66827a3a424d1f743).

## Detailed context
Apparently there is a problem with retrieved encoder velocity or encoder values.
It seems that there is a `-1` missing somewhere (sign mismatch) between the motor values and velocities. Looking at the patterns of motor value (`motor val fb`) and velocity feedbacks `motor vel fb`, when the motor values are decreasing the velocity is positive and vice versa.
Instead, when I compute the motor values (`motor val comp`) from the motor velocity feedbacks or when I compute the velocities (`motor vel comp`) from the motor value feedback the computed ones seems to be coherent with the encoder feedback that I have used.

## Accompanying material
This figure are plotted for the thumb motor associated with the distal joint (`r_thumb_distal`).
[image]

[image]

The link to my committed implementation robotology/walking-teleoperation@3a3dc78

cc @pattacini  @traversaro

pattacini, 2020-10-31:
Thanks for reporting @kouroshD 👍🏻 
We'll be giving a look at it in the next days.

Nicogene, 2020-11-02:
Hi @kouroshD!

is it possible to have a dump of the `stateExt:o` of both arms?
And which fw is used for the board of the hands?

kouroshD, 2020-11-02:
I have not dumped the data related to `stateExt:o`, I have the two following datasets if you want to use them and the matlab scripts to read those data and in order to plot the data. Only make unzip and copy them in the same directory, and make a folder with the datasets name (since, it save the plots!)

[HapticGloveModule_2020-10-29_11-03-37_log.mat.zip](https://github.com/robotology/icub-tech-support/files/5476529/HapticGloveModule_2020-10-29_11-03-37_log.mat.zip)

[HapticGloveModule_2020-10-29_11-20-58_log.mat.zip](https://github.com/robotology/icub-tech-support/files/5476532/HapticGloveModule_2020-10-29_11-20-58_log.mat.zip)

[PlotData.m.zip](https://github.com/robotology/icub-tech-support/files/5476535/PlotData.m.zip)

Regarding the fw version, I am not sure about it, probably the last version! But this is sth to check.

kouroshD, 2020-11-02:
Plotting the data of the first motor of the thumb, I have noticed that its velocity feedback value are zero.
[image]

kouroshD, 2020-11-03:
Thanks to @gabrielenava , we have the following dataset and info regarding the fw versions.

[image]

[stateExt_00003.zip](https://github.com/robotology/icub-tech-support/files/5483548/stateExt_00003.zip)

cc @Nicogene

Nicogene, 2020-11-05:
Thanks @kouroshD and @gabrielenava !

Nicogene, 2020-11-06:
I analyzed the data using a modified version of this [script](https://github.com/dic-iit/MAPest/blob/master/src/readStateExt.m)
<details>
  <summary>Click to expand!</summary>
  
~~~matlab
% Copyright (C) 2017 Istituto Italiano di Tecnologia (IIT)
% All rights reserved.
%
% This software may be modified and distributed under the terms of the
% GNU Lesser General Public License v2.1 or any later version.

function [q, dq, d2q, m, dm, d2m, time] = readStateExt(n, filename)
% READSTATEEXT reads data from robot acqisizion. 
%
% Inputs 
% - n       : is the number of degrees of freedom of the part of the robot;
% - filename: is the name of the file containing the data from stateEXT:o 
%             port.

format = '%d %f ';
fid    = fopen(filename);

data   = textscan(fid,'%s','delimiter','\n');
text=data{1};

nlines = length(text);
q=[];
dq=[];
d2q=[];
m=[];
dm=[];
d2m=[];
for j = 1 : 10
   format = [format, '('];
   for i = 1 : n
      if j < 9
         format = [format, '%f '];
      else
         format = [format, '%d '];
      end
   end
   format = [format, ') [ok] '];
end
for i=1:nlines
    line = text{i};
    C    = textscan(line, format);
    %size(C)
    time = C{1, 2};
    %size(cell2mat(C(1, 3    :3+  n-1)))
    %size(q')
    q    = [q; cell2mat(C(1, 3    :3+  n-1))]; % n columns of "q" value
    dq   = [dq; cell2mat(C(1, 3+  n:3+2*n-1))]; % n columns of "dq" value
    d2q  = [d2q; cell2mat(C(1, 3+2*n:3+3*n-1))]; % n columns of "d2q" value

    m    = [m; cell2mat(C(1, 3+3*n:3+4*n-1))]; % n columns of "m" value
    dm   = [dm; cell2mat(C(1, 3+4*n:3+5*n-1))]; % n columns of "dm" value
    d2m  = [d2m; cell2mat(C(1, 3+5*n:3+6*n-1))]; % n columns of "d2m" value

end

if fclose(fid) == -1
   error('[ERROR] there was a problem in closing the file')
end
end
~~~
</details>

The original script seemed to read only one line and read only q, dq, and d2q(a.k.a encoder position, velocity, and acceleration)

My modified version read the whole file and reads also m, dm, and d2m, that should be motor position, velocity, and acceleration(according to (https://github.com/robotology/yarp/blob/master/src/libYARP_dev/src/idl_generated_code/yarp/dev/impl/jointData.cpp#L180L216)

(Is it the case to open a PR for adding this version? Can be useful? cc @traversaro)

I found the same strange behavior of fingers that @kouroshD  found also on the motor side.

|   | `thumb_oppose` | `thumb_distal`  | 
|---|---|---|
| pos_encoder  | [image]  |  [image]  |  
|  pos_motor |  [image] |  [image] |
| vel_encoder  |  [image]  | [image]  |
| vel_motor  |  [image]  |  [image]  | 
 

I hope that this analysis can give hints to @ale-git in order to see the source of the problem

Here is the resulting workspace
[testFingersProblem.zip](https://github.com/icub-tech-iit/fix/files/5495978/testFingersProblem.zip)

traversaro, 2020-11-06:
> (Is it the case to open a PR for adding this version? Can be useful? cc @traversaro)

Yes sure, feel free to propose a PR, I suspect it will not be used in MAPEst itself, but that script is frequently copied around, so it is good to have a single point where to report improvements.

pattacini, 2020-11-07:
@ale-git @kouroshD @Nicogene 
Essentially:
- Thumb distal: although the motor and the encoder positions do have opposite signs, their velocities do have the same sign.
- Thumb opposition: velocities are somehow not computed.

kouroshD, 2020-11-07:
@pattacini  If I am not wrong, this is the behavior for all the hand motors.

Nicogene, 2020-11-09:
Hi @kouroshD !

@ale-git says:

>About the thumb distal, my best guess is that the Gearbox_M2J parameter has the wrong sign.
The parameter is in https://github.com/robotology/robots-configuration/blob/devel/iCubGenova04/hardware/mechanicals/right_arm-eb28-j8_11-mec.xml
About the thumb opposition I think that the motor quadrature encoder could be simply broken.
I suggest a quick test by changing the thumb distal Gearbox_M2J sign in the configuration file linked above.

Could you please try to change the Gearbox_M2J parameter and try again?

kouroshD, 2020-11-10:
Thank you @Nicogene  for the updates.

> About the thumb opposition I think that the motor quadrature encoder could be simply broken.

I will try with the left hand.

> I suggest a quick test by changing the thumb distal Gearbox_M2J sign in the configuration file linked above.

So, you are suggesting to reverse the sign of the `Gearbox_M2J` values written here https://github.com/robotology/robots-configuration/blob/d415b08661e9491b0193ecdcb38a460bf6744177/iCubGenova04/hardware/mechanicals/right_arm-eb28-j8_11-mec.xml#L16, from `256` to `-256`?!

kouroshD, 2020-11-11:
> About the thumb opposition I think that the motor quadrature encoder could be simply broken.

 @Nicogene  Just to understand better, the velocity feedback is computed by time differentiation of the encoder values, right? Since we have values for `thumb_oppose`, then is the broken motor quadrature encoder a valid assumption?!

pattacini, 2020-11-11:
>@Nicogene Just to understand better, the velocity feedback is computed by time differentiation of the encoder values, right? Since we have values for thumb_oppose, then is the broken motor quadrature encoder a valid assumption?!

@kouroshD made a good point indeed. Any idea @ale-git @Nicogene ?

Changing the sign of `Gearbox_M2J` doesn't sound unusual as there are other cases where the gear ratio is inverted to take into account opposite rotations of the two pulleys. See for example https://github.com/robotology/robots-configuration/search?q=Gearbox_M2J.

Nicogene, 2020-11-11:
>So, you are suggesting to reverse the sign of the Gearbox_M2J values written here https://github.com/robotology/robots-configuration/blob/d415b08661e9491b0193ecdcb38a460bf6744177/iCubGenova04/hardware/mechanicals/right_arm-eb28-j8_11-mec.xml#L16, from 256 to -256?!

Yes if you search this param in `robots-configuration` sometimes it is set with the - probably for compensating issues like this.
A test on the other arm will be exahaustive in this sense, I expect that Gearbox_M2J has to have the same sign between arms

> @Nicogene Just to understand better, the velocity feedback is computed by time differentiation of the encoder values, right? Since we have values for thumb_oppose, then is the broken motor quadrature encoder a valid assumption?!

The encoder velocity is computed as follows https://github.com/robotology/icub-firmware/blob/master/emBODY/eBcode/arch-arm/embobj/plus/mc/AbsEncoder.c#L480-L508, actually yes it seems that it is computed by time differentiation but maybe @ale-git can correct me.

kouroshD, 2020-11-12:
Yesterday, I have collected two datasets, one is related to logging the data of both arms left and right, without changing any parameter. I moved hand motors plus wrist and elbow motors. The data is in the following:
[loggedfileWithoutChanges.zip](https://github.com/robotology/icub-tech-support/files/5529207/loggedfileWithoutChanges.zip)

Following your suggestion, I have changed the sign of Gearbox_M2J of the right hand (the ones in [this link](https://github.com/robotology/robots-configuration/blob/d415b08661e9491b0193ecdcb38a460bf6744177/iCubGenova04/hardware/mechanicals/right_arm-eb28-j8_11-mec.xml#L16) to `-256`), and here is the logged data:
[loggedAfterChanging Gearbox_M2J.zip](https://github.com/robotology/icub-tech-support/files/5529223/loggedAfterChanging.Gearbox_M2J.zip)

ale-git, 2020-11-12:
@xEnVrE has found this comment:

```
        /*
        NOTE: (VALE)
            In this board, thumb oppose and thumb proximal joints are coupled.
            I get decoupled matrix from mc4(CAN) firmware version 0x228 function "decouple_dutycycle".
            In this function, the fw uses a different decoupled matrix if it runs on left arm or right arm.
            Currently I tested only left arm, so we need to check if the same matrix can be use also on twin board on right arm.
        */
```

This part is not used any more because the coupling matrices have been moved to the configuration files, but if there was an error in the right hand it has been moved to the configuration file as well, since the values are the same for right and left hands.

https://github.com/robotology/icub-firmware/blob/f5850b0d935e814eab5f1a448a6471b141dfcc5f/emBODY/eBcode/arch-arm/embobj/plus/mc/Controller.c#L1135

ale-git, 2020-11-12:
I didn't know that there is a coupling in the thumb. I'll check the mc4 firmware in order to see if the values are different for the two hands.

Nicogene, 2020-11-12:
I analyzed the data, as before on the right are reported the plots of `thumb_distal`, on the left the `thumb_oppose`.

Here is the plot regarding the right arm **with the M2J fix**.

[image]

As expected now motor velocity and encoder velocity now has opposite signs, and now the encoder velocity is coherent with the encoder values.

Here is the data **before the changes** for left arm and right arm respectively

[image]
[image]

pattacini, 2020-11-12:
So we managed to solve the problem related to the sign of the velocity, cool 👍🏻 
The velocity of the thumb opposition is still zero though.

pattacini, 2020-11-12:
Hi @kouroshD 

Do you mind opening up a PR to fix the sign problem?
Meanwhile, we'll be reasoning on the thumb opposition problem.

kouroshD, 2020-11-13:
> Hi @kouroshD
> 
> Do you mind opening up a PR to fix the sign problem?
> Meanwhile, we'll be reasoning on the thumb opposition problem.

Sure, I will do it. @pattacini  @Nicogene Thank you for the investigation that and spotting the problem.

pattacini, 2020-11-13:
Concerning the velocity of the thumb opposition, @ale-git is investigating whether the following explanation may be plausible.

The coupling matrices (direct and inverse) contain the correct signs that give the direction of motion but wrong absolute values. This hypothesis may explain why the position control is behaving as expected (we just need the correct signs) and why, by contrast, the computed velocities end up being very small (practically equal to zero).

Investigating... ⚙

ale-git, 2020-11-19:
The source of the hand coupling model is the CAN robot, in particular:

https://github.com/robotology/icub-firmware/blob/60c6b0485d178a08f124e000212239fc32ddd51b/motorControllerDsp56f807/common_source_code/pwm_decoupling.c#L126

```
    if (_kp[0] > 0)
    {
        //right hand
        pwm_out[0] = pwm[0];
        pwm_out[1] = pwm[1] + pwm[0];
        pwm_out[2] = pwm[2];
        pd_out[0] = _pd[0];
        pd_out[1] = _pd[1] + _pd[0];
        pd_out[2] = _pd[2];
    }
    else
    {
        //left hand
        pwm_out[0] = pwm[0];
        pwm_out[1] = pwm[1] - pwm[0];
        pwm_out[2] = pwm[2];
        pd_out[0] = _pd[0];
        pd_out[1] = _pd[1] - _pd[0];
        pd_out[2] = _pd[2];
    }
```

In the .xml configuration file hardware/mechanicals/right_arm-eb28-j8_11-mec.xml we have:

```
    <group name="COUPLINGS"> 
                            
        <param name ="matrixJ2M"> 
            1.000   0.000   0.000   0.000
           -1.000  +1.000   0.000   0.000
            0.000   0.000   1.000   0.000
            0.000   0.000   0.000   1.000   
        </param>

         <param name ="matrixM2J"> 
            1.000   0.000   0.000   0.000
           +1.000  +1.000   0.000   0.000
            0.000   0.000   1.000   0.000
            0.000   0.000   0.000   1.000   
        </param>
```

It seems wrong to me since it is identical to the left hand. My guess is that the right configuration file has been copied from the left one but the coupling numbers haven't been updated. Accordingly to the CAN robot originals, the coupling in the right hand should be:

```
    <group name="COUPLINGS"> 
                            
        <param name ="matrixJ2M"> 
            1.000   0.000   0.000   0.000
           +1.000  +1.000   0.000   0.000
            0.000   0.000   1.000   0.000
            0.000   0.000   0.000   1.000   
        </param>

         <param name ="matrixM2J"> 
            1.000   0.000   0.000   0.000
           -1.000  +1.000   0.000   0.000
            0.000   0.000   1.000   0.000
            0.000   0.000   0.000   1.000   
        </param>
```

pattacini, 2020-11-19:
Thanks heaps @ale-git 👍🏻 

I believe this is only part of the investigation yet, which should allow us to sort out a bug anyway, as it's unlikely that this can explain why we don't get velocities from both hands.

pattacini, 2020-11-19:
@kouroshD I would ask you to integrate into the upcoming PR also the [modifications][1] highlighted by @ale-git above in order to fix the control of the right thumb, at least.

Once done, could you please report here the outcome of any ongoing test?

We talked to @ale-git today and he also agrees that there must be still something else on the velocity side.

Still investigating... ⚙

[1]: https://github.com/robotology/icub-tech-support/issues/1019#issuecomment-730148256

kouroshD, 2020-11-24:
@pattacini I should open the PR on `devel`, right?!

pattacini, 2020-11-24:
I'd say so yes, thanks! 👍🏻 

@Nicogene let's keep an eye wide open on this because of the upcoming distro 2020.11 👀

kouroshD, 2020-11-24:
I've opened the PR in https://github.com/robotology/robots-configuration/pull/206

I've added the fixes for all the  motors of the left and right hands. Moreover, I've added the fix suggested in https://github.com/robotology/icub-tech-support/issues/1019#issuecomment-730148256

pattacini, 2020-11-24:
Thanks a lot @kouroshD 👍🏻 
PR merged ✔️ 

I would keep still open anyway to address https://github.com/robotology/icub-tech-support/issues/1019#issuecomment-730253787.

kouroshD, 2020-11-24:
Thanks a lot @pattacini

pattacini, 2020-11-24:
@kouroshD did you have already the chance to test these modifications?

kouroshD, 2020-11-24:
> @kouroshD did you have already the chance to test these modifications?

No, not yet. Others were using the robot, and there is a problem with robot configuration installation to be investigated.
cc @traversaro

pattacini, 2020-11-26:
Hi @kouroshD 

Regarding the null velocity, I had the chance to talk to @ale-git today, who told me that velocities seem to be computed by differentiating the motor encoders. Look at https://github.com/robotology/robots-configuration/blob/master/iCubGenova04/hardware/mechanicals/left_arm-eb26-j12_15-mec.xml#L18.

This is conceptually wrong. Therefore, you could proceed by trying to change those parameters to all zeros. This way, the differentiation will be applied to the joint encoders, as it should be.

Once done, you could open up another convenient PR.

The fact that the velocity of the thumb opposition is zero stems then from the motor encoder that remains unvaried as well (probably it's broken).

Awaiting your feedback on this.

kouroshD, 2020-11-26:
OK, thanks to @traversaro,  we fixed the problem mentioned here. But then, the right hand of iCub is sort of broken so, I cannot yet confirm for that. However, I did the check for the left hand. It seems like not all the values should become `-`. So, here is the result for all the left-hand motors.

|  Left hand values before the fix | Left hand values after the  fix  |
|---|---|
| [image] |  [image] |
|  [image] |  [image] |
| [image]   |  [image] |
|  [image] |  [image] |
| [image] |  [image] |
| [image]  | [image]  |
| [image]  |  [image] |
|  [image]  |  [image]  |

The dashed lines are computed not feedbacks. 
- Blue line: motor angle feedback
- Blue dashed line: motor angle computed from integration of velocities
- Orange line: motor velocity feedback
- Orange dashed line: motor velocities computed from derivation of motor angles

###
So, the values should for the left hand int configuration files should be:
```
Motor 1:  X         Motor 5:  -256
Motor 2:  256    Motor 6:    256
Motor 3:  256    Motor 7:    256
Motor 4:  256    Motor 8:    -256
```
Data for before changing the configuration values:
[Archive.zip](https://github.com/robotology/icub-tech-support/files/5604064/Archive.zip)

Data for before after changing the configuration values:
[Archive 2.zip](https://github.com/robotology/icub-tech-support/files/5604068/Archive.2.zip)

I will fix this for the next PR.

pattacini, 2020-11-26:
Great debugging @kouroshD 👍🏻 

>It seems like not all the values should become `-`

You know why? To fix a wrong winding of the cables around the pulley in the past it was common to change the sign of those parameters. This shouldn't be happening any longer (hopefully).

kouroshD, 2020-11-26:
> Hi @kouroshD
Regarding the null velocity, I had the chance to talk to @ale-git today, who told me that velocities seem to be computed by differentiating the motor encoders. Look at https://github.com/robotology/robots-configuration/blob/master/iCubGenova04/hardware/mechanicals/left_arm-eb26-j12_15-mec.xml#L18.
This is conceptually wrong. Therefore, you could proceed by trying to change all the values of useMotorSpeedFbk to zero. This way, the differentiation will be applied to the joint encoders, as it should be.

So, this means, if `useMotorSpeedFbk` is to true, then for force feedback we are using the motor encoders, otherwise we are using the joint encoders to compute the velocity feedback.

>  Once done, you could open up another convenient PR.
The fact that the velocity of the thumb opposition is zero stems then from the motor encoder that remains unvaried as well (probably it's broken).
Awaiting your feedback on this.

Yes, I will do the requested changes. As you said it can be because of broken motor encoders in that case. However, the two thumbs together have the same problem has low probability. I will do it tonight and let you know.

pattacini, 2020-11-26:
> Do you think the problem of wrong winding can be addressed in that issue as well. If not, I will keep the values we found. If this can be addressed as well, we can keep all the values equal to 256.

Rewinding cables can be quite time-consuming, I'm afraid. Check that with @julijenv, but I believe we can survive for the time being with the current situation.

Essentially, software-wise we had to:
- update the calibration matrices
- fiddle with `Gearbox_M2J`
- put `useMotorSpeedFbk` equal to 0

You applied/are going to apply those changes to `iCubGenova04`. Can you double-check that?

Then, we'll be dealing with extending the mods to all remaining robots.

kouroshD, 2020-11-27:
> Rewinding cables can be quite time-consuming, I'm afraid. Check that with @julijenv, but I believe we can survive for the time being with the current situation.

OK, I'll check with him, otherwise I'll find the signs for `Gearbox_M2J` of the right hand after it is fixed.

> update the calibration matrices

Which calibration matrix? you mean coupling `matrixJ2M ` matrix? If yes, I've done that before!

> put useMotorSpeedFbk equal to 0

I should double-check this for the right arm, but currently is broken. 

@pattacini  @traversaro  As a matter of curiosity, I've checked all the values of `useMotorSpeedFbk`for all the robot parts and they are equal to `1`. Then, how we control the robot at higher levels if somebody needs the velocity feedback?!  I mean for some motor/joints part may not change if there is no coupling there, but for the coupled motor/joints it should change a lot? Shoudn't we put `useMotorSpeedFbk` for all the robot by default to `0`?

pattacini, 2020-11-27:
>Which calibration matrix? you mean coupling matrixJ2M matrix? If yes, I've done that before!

Yep, sorry, coupling matrix. I've amended my comment above.

> @pattacini @traversaro As a matter of curiosity, I've checked all the values of useMotorSpeedFbkfor all the robot parts and they are equal to 1. Then, how we control the robot at higher levels if [...]

We are going to update all the robots for what concerns the finger control, where actually the velocity feedback is never used, or at least the velocities are usually computed at the high level.

The fact that the motor joints are always chosen for doing differentiation I think stems from the high-speed (optical) encoders that provide much more ticks per round in the hope that this gives in turn more stable estimates for the speed. I've never been convinced about that, honestly. As you can see from your plots, differentiating high-speed encoders does not guarantee clean velocity signals.

traversaro, 2020-11-27:
fyi @isorrentino (in particular the last comment by @pattacini)

kouroshD, 2020-11-27:
@pattacini  But then, in this case the coupling between the motors and joints should be taken into account for the higher level control, no?! Because at the end what is important for higher level controllers are the joint velocities? For example consider the shoulder joints that the motors and joints are coupled.

cc @DanielePucci

pattacini, 2020-11-27:
> @pattacini But then, in this case the coupling between the motors and joints should be taken into account for the higher level control, no?! Because at the end what is important for higher level [...]

Sure, but the velocities computed at the motor side are then of course processed through the coupling and/or simply the gear ratio and reported to the joint side. By doing this, we're assuming that the map remains constant, which is actually a good approximation of a tendon-driven actuation, but you know, not really for the hands where we have slack and a lot of felixibility.

kouroshD, 2020-11-27:
As soon as we have the right hand again, I will check that, and open a PR.

Uboldi80, 2022-04-20:
Following a thorough check on the hands that are currently on my desk for maintenance, we found that some proximal tendons were wrapped incorrectly, moreover almost certainly, as indicated by @gsisinna , following the profound experience made on the joints of the inches and on the coupling of the same, I imagine that most likely the coupling matrix is to be verified.

closed 2022-09-01 by pattacini