robotology/mesh-support #1020: iCubGenova08: Right arm J3 hw fault

ddetommaso, 2020-11-02:
## Brief description of the request/failure
The elbow of the right arm stops working due to an hardware fault on J3.

## Detailed context
[image]

## Accompanying material
[log_pc104_yarprobotinterface_1579.txt](https://github.com/robotology/icub-tech-support/files/5473727/log_pc104_yarprobotinterface_1579.txt)

maggia80, 2020-11-02:
@julijenv  @violadelbono  please check here the docs: [iCub 2.7 Logic&Harness](https://icub-tech-iit.github.io/documentation/icub_wiring/icub2_x/#logic-and-harness-icub-27-e36-full-robot-with-high-performance-imu-on-the-waist-and-new-face-expressions)

the faulty encoder should be the one in the picture below:  
[image]

julijenv, 2020-11-02:
Hi @ddetommaso , As mentionned previously by @maggia80 (thx btw) both the yarpmotorgui and the lof file direct to this problem.
I will try to schedule an intervnetion, I will try to get the car to pass by. Which day for you would be better in the week in order to do not disturb your planning?
let us know, ASAP, when you give me some day I will ask the car and in the same hope to get feedback in order to let you know

ddetommaso, 2020-11-02:
Hi @julijenv 

thank you for your availability.
In BP a suitable time schedule would be on Friday starting from 11:00. Please contact directly D. Ghiglino  for further information.

julijenv, 2020-11-06:
Hi there,
the cable ground 
[image]

was barely connected , which was actually creating the fault from time to time depending on th e movement.
recrimping save the issue.
we made almost 70 trials with 0% failure so I close the issue

closed 2020-11-06 by julijenv