robotology/mesh-support #1021: MC: exernal fault button pressed.

tKsome, 2020-11-03:
## Brief description of the request/failure
the power has been opened but we cannot open yarprobotinterface.  we have call you via Skype. You can get more infomation in there.

## Detailed context

## Accompanying material

julijenv, 2020-11-03:
Hi @tKsome ,
It is always necessary to provide the log file in order to try to understand the problem you are facing.
Calling us via Skype is NOT the entry point for getting in touch with us, we need you to fill up an issue in order to receive feedback from us and knowing that you are facing a problem.
If ever your problem can be resolved by helping you via SKype we surely do. But we also have plenty of things to do so we are not 100% of our time behind SKype.
It is our duty to help you out but we need to organize ourselves to be reachable on SKype. 
At the moment as requested on any open issue, please send us the log file in order to point out the problem.
thx for your comprehension,

BTW: `MC: exernal fault button pressed` means that the red fault button is pressed, R u sure that the jack of the red fault button is properly plugged and of course that is NOT pressed.

waiting to get your log file, thx in advacne

tKsome, 2020-11-03:
owww sorry , we have got the solve

julijenv, 2020-11-03:
Hi @tKsome , could you please explain what was the problem you found and resolved?
This issue can be read by other people and if it can be helpful for them, at least they know what has been done.
after explaining, could you please close the issue .
thx in advacne

closed 2020-12-30 by julijenv