robotology/mesh-support #1023: [FT Shoes]: Faulty Cable

CarlottaSartore, 2020-11-05:
## Brief description of the request/failure
While I and @lrapetti  were performing some experiments with the FT Shoes, we had few problems with one of the canbus cable that is not working properly (the one identified by the pink "tape" as you can see in the picture attached) and we obtain the error that you can see in attachment. 

The only configuration that works is the following: 

1) One shoe connected via the CanBus with the faulty cable (using the not faulty one) to one PC; 

2) One shoe connected via another CanBus (that has got only one cable) to another PC; 

C.C. @violadelbono  @claudia-lat  @Yeshasvitvs 
## Accompanying material
[image]

error message 
```

PS C:\Users\icub\dev\wearables\app\xml> yarprobotinterface.exe --config .\FTShoeRightWearableDevice.xml
||| configuring
||| default config file specified as yarprobotinterface.ini
||| checking [C:\Users\icub\dev\wearables\app\xml\yarprobotinterface.ini] (pwd)
||| checking [C:\Users\icub\AppData\Roaming\yarp\config\yarprobotinterface.ini] (YARP_CONFIG_HOME)
||| checking [C:\Users\icub\AppData\Roaming\yarp\yarprobotinterface.ini] (YARP_DATA_HOME)
||| checking [C:\ProgramData\yarp\yarprobotinterface.ini] (YARP_CONFIG_DIRS)
||| checking [C:\Users\icub\dev\robotology-superbuild\build\install\share\retargeting-robot-models\yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [C:\Users\icub\dev\robotology-superbuild\build\install\share\human-gazebo\yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [C:\Users\icub\dev\robotology-superbuild\build\install\share\HDERviz\urdfs\yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [C:/Users/icub/dev/robotology-superbuild/build/install/share/codyco\yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [c:\Users\icub\dev\robotology-superbuild\robotology\icub-tests\suits\yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [c:\Users\icub\dev\robotology-superbuild\build\install\share\ICUBcontrib\yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [c:\Users\icub\dev\robotology-superbuild\build\install\share\iCub\yarprobotinterface.ini] (YARP_DATA_DIRS)
||| checking [c:\Users\icub\dev\robotology-superbuild\build\install\share\yarp\yarprobotinterface.ini] (YARP_DATA_DIRS)
obotology-superbuildbuildinstallshareICUBcontrib\yarprobotinterface.ini] (yarp.d)
obotology-superbuildbuildinstallshareiCub\yarprobotinterface.ini] (yarp.d)
||| did not find yarprobotinterface.ini
yarp: cannot read from yarprobotinterface.ini
||| finding file [config]
||| checking [C:\Users\icub\dev\wearables\app\xml\.\FTShoeRightWearableDevice.xml] (pwd)
||| found C:\Users\icub\dev\wearables\app\xml\.\FTShoeRightWearableDevice.xml
[DEBUG]Reading file C:\Users\icub\dev\wearables\app\xml\.\FTShoeRightWearableDevice.xml
[DEBUG]yarprobotinterface: using xml parser for DTD v3.x
[DEBUG]Reading file C:\Users\icub\dev\wearables\app\xml\.\FTShoeRightWearableDevice.xml
[DEBUG]Preprocessor complete in:  2.91783e-06 s
||| finding paths [plugins]
||| checking [C:\Users\icub\dev\wearables\app\xml\plugins] (pwd)
||| checking [C:\Users\icub\AppData\Roaming\yarp\config\plugins] (YARP_CONFIG_HOME)
||| checking [C:\Users\icub\AppData\Roaming\yarp\plugins] (YARP_DATA_HOME)
||| checking [C:\ProgramData\yarp\plugins] (YARP_CONFIG_DIRS)
||| checking [C:\Users\icub\dev\robotology-superbuild\build\install\share\retargeting-robot-models\plugins] (YARP_DATA_DIRS)
||| checking [C:\Users\icub\dev\robotology-superbuild\build\install\share\human-gazebo\plugins] (YARP_DATA_DIRS)
||| checking [C:\Users\icub\dev\robotology-superbuild\build\install\share\HDERviz\urdfs\plugins] (YARP_DATA_DIRS)
||| checking [C:/Users/icub/dev/robotology-superbuild/build/install/share/codyco\plugins] (YARP_DATA_DIRS)
||| checking [c:\Users\icub\dev\robotology-superbuild\robotology\icub-tests\suits\plugins] (YARP_DATA_DIRS)
||| checking [c:\Users\icub\dev\robotology-superbuild\build\install\share\ICUBcontrib\plugins] (YARP_DATA_DIRS)
||| found c:\Users\icub\dev\robotology-superbuild\build\install\share\ICUBcontrib\plugins
||| checking [c:\Users\icub\dev\robotology-superbuild\build\install\share\iCub\plugins] (YARP_DATA_DIRS)
||| found c:\Users\icub\dev\robotology-superbuild\build\install\share\iCub\plugins
||| checking [c:\Users\icub\dev\robotology-superbuild\build\install\share\yarp\plugins] (YARP_DATA_DIRS)
||| found c:\Users\icub\dev\robotology-superbuild\build\install\share\yarp\plugins
obotology-superbuildbuildinstallshareICUBcontrib\plugins] (yarp.d)
obotology-superbuildbuildinstallshareiCub\plugins] (yarp.d)
yarp: Port //yarprobotinterface active at tcp://127.0.0.1:10002/
[INFO]startup phase starting...
||| finding paths [plugins]
(canAddress 3) (canDeviceNum 0) (canbusDevice ecan) (channels 6) (device canBusAnalogSensor) (diagnostic 1) (format 16) (period 5) (physDevice ecan) (robotName) (useCalibration 1)
||| finding paths [plugins]
[INFO]created device <ecan>. See C++ class yarp::dev::EsdCan for documentation.
[DEBUG]using broadcast period 5 on device
[DEBUG]Using internal calibration on device
[ERROR]Trying to get fullscale data from sensor 3 net [0]: no answer received or message lost (ch:0)
[ERROR]Trying to get fullscale data from sensor 3 net [0]: no answer received or message lost (ch:0)
[... 307 lines cut ...]
[[ERRORERROR]]AnalogWrapper: AnalogServer: Sensor returned timeout error (code 3).AnalogWrapper: AnalogServer: Sensor returned timeout error (code 3).

[ERROR[]AnalogWrapper: AnalogServer: Sensor returned error with code 29231.[
yarp: Port /FTShoeRight/WearableData/metadataRpc:o active at tcp://127.0.0.1:10010/
ERROR]ERROR]AnalogWrapper: AnalogServer: Sensor returned timeout error (code 3).[DEBUGAnalogWrapper: AnalogServer: Sensor returned timeout error (code 3).]IWearWrapper : attach() successful

[INFO]All actions for action level 5 of startup phase started. Waiting for unfinished actions.
[[INFO]All actions for action level 5 of startup phase finished.
[ERROR]ERROR[AnalogWrapper: AnalogServer: Sensor returned error with code 29231.[]ERRORAnalogWrapper: AnalogServer: Sensor returned timeout error (code 3).]AnalogWrapper: AnalogServer: Sensor returned timeout error (code 3).WARNING

]
There was some problem running actions for startup phase . Please check the log and your configuration
[INFO]startup phase finished.
[ERROR]Error in startup phase... see previous messages for more info
[WARNING]Interrupt # 1 # received.
[INFO]Interrupt received. Stopping all running threads.
[INFO]interrupt1 phase starting...
[INFO]interrupt1 phase finished.
[INFO]shutdown phase starting...
[INFO]Entering action level 5 of phase shutdown
[INFO]All actions for action level 5 of shutdown phase started. Waiting for unfinished actions.
[INFO]All actions for action level 5 of shutdown phase finished.
[INFO]shutdown phase finished.
[INFO]RFModule failed to open.
```

julijenv, 2020-11-09:
everthig has been repaired, could you please let me know if it works?
@lrapetti , @CarlottaSartore - if it is ok, please close the issue , thx in advance

CarlottaSartore, 2020-11-09:
We did some experiments with the FT shoes today and everything went fine, Thanks @julijenv ! 
Closing the issue

closed 2020-11-09 by CarlottaSartore