robotology/mesh-support #1024: iCubGenova01 left wrist and hand_finger faults/idles at startup

xEnVrE, 2020-11-05:
## Brief description of the request/failure
When starting up the robot, some joints of the left arm are in fault/idle.

I attach a picture to summarize the situation.

[image]

By idling and running the parts everything seems to be ok, but still at every startup the fault occurs.

Please also find attached the log from the `yarprobotinterface`.

## Accompanying material

[log_pc104_yarprobotinterface_1546.txt](https://github.com/robotology/icub-tech-support/files/5496060/log_pc104_yarprobotinterface_1546.txt)

julijenv, 2020-11-19:
Ciao @xEnVrE ,
I just pass 3 hours debugging your robot, but unfortunately the first problem on which I ran into was that the board BLL form the right arm was not showing up, after having make a nice debug and repair the stadoff- in the back and doulbe every wiring, Now the BLL are Ok.
Later on I try to launch the robot and I didnt have any problem with the wirst. So I will let this opend until it reoccured, please let me know

xEnVrE, 2020-11-19:
Hi @julijenv 

thank you so much for the debugging. My feeling is that there were problems that sometimes do not show off because of loose wiring. Happy to know that BLLs are now ok.

I'll tell you if the problem with the wrist occurs again.

xEnVrE, 2020-11-20:
Hi @julijenv,

this morning the issue occurred again. Precisely, after startup I see the following from the `yarpmotorgui`:

[image]

Taking a look at the log, it seems that something is going wrong during the calibration of the group of joints containing the joint 6, i.e. `l_wrist_yaw`

```
132,218766 <ERROR> Left_Arm_Calibrator : checkGoneToZeroThreshold: joint  6  is idle, skipping! 
 132,228741 <DEBUG> Left_Arm_Calibrator: checkGoneToZeroThreshold: joint: 7 curr: 15.756 des: 15.000 -> delta: 0.756 threshold: 90.000 output: 0.000 mode: pos 
 132,228891 <DEBUG> Left_Arm_Calibrator: checkGoneToZeroThreshold: joint: 7 completed with delta: 0.756 over: 90.000 
 132,228952 <ERROR> Left_Arm_Calibrator : set 2 : some axis got timeout while reaching zero position... disabling this set of axes 
 132,238907 <DEBUG> sharedcan [1] msg from board 3: PWM DIS:0  
 132,239052 <ERROR> sharedcan [1] board 3 FAULT OVERLOAD AXIS 0 
```

Please also find attached the log from the `yarprobotinterface`.

[log_pc104_yarprobotinterface_1262.txt](https://github.com/robotology/icub-tech-support/files/5572704/log_pc104_yarprobotinterface_1262.txt)

julijenv, 2020-12-10:
I close it because I think we made it ok last week ricalibrating the 2 wrists

xEnVrE, 2021-02-19:
Hi @julijenv,

the issue is still happening unfortunately.

julijenv, 2021-02-19:
another insights given by @xEnVrE  is that the calibration actually stop when the pronosuppination goes to the limit which tensds to mean that the power on the board may drop , it would be nice to check the 8 pins connector of the the power line of this board.

xEnVrE, 2021-06-13:
The issue is not presenting anymore, we can close for now.

closed 2021-06-13 by xEnVrE