robotology/mesh-support #1025: iCubLausanne02: invalid data on left knee Magnetic encoder and Torso starting in idle mode

MichaelBombile, 2020-11-06:
Hi @julijenv ,

We have a hardware problem with the left knee, which goes in Hardware fault mode. It complains about invalid data from the AEA magnetic encoder : 
``` [ERROR] from BOARD 10.0.1.6 (left_leg-eb6-j0_3), src LOCAL, adr 0, time 147s 351m 521u: (code 0x02000008, par16 0x0003 par64 0x0000000000000002) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + . ```  

And also an error at the torso joints which start and stay in the idle mode. The error thrown is : ``` [ERROR] from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 138s 312m 205u: (code 0x02000000, par16 0x0002 par64 0x0000000000000000) -> MC: exernal fault button pressed. + ``` . This is not the case since the other robot joints except the left knee start normally.

In reference to similar issue (https://github.com/robotology/icub-tech-support/issues/302), I tried to check all the sensors cables connections, all seems fine and now I don't know exactly for to look for.
Could you please assist us in this regards. Find attached the entire log file
[iCubLausanne02_log_06112020_1101.txt](https://github.com/robotology/icub-tech-support/files/5500755/iCubLausanne02_log_06112020_1101.txt)

Thanks in advance.

julijenv, 2020-11-09:
Hi @MichaelBombile ,
FRom your log file and your perfect comprehensin of the error messages, you identify correctly, the problems occuring in your robot.
the first one :
`[ERROR] from BOARD 10.0.1.6 (left_leg-eb6-j0_3), src LOCAL, adr 0, time 147s 351m 521u: (code 0x02000008, par16 0x0003 par64 0x0000000000000002) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .`
to solve the problem with this paticular one you need to take a look at the level of the left knee.
i dont know if you have a robot wih backpack or not but the folowing file

https://github.com/icub-tech-iit/electronics-public/blob/master/projects/robots/iCub2.5/iCub%20wiring%20Rev3.3%20COM-EXP%20and%20MC4-PLUS%20and%20BatteryPack/pdf/Motor%20and%20board%20placement_iCub2.5.pdf 

should show wher are located both motor and encoder al over the body for ETH based robot.
[image]

bascially your error message is due to a broken cable for this specific sensor either at the level of the board - external knee - or at the level of the EMS (controlling board - the connector attached to this board) has one or more cable broken, check and repair and you should be ok then.

For the problem:
`[ERROR] from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 138s 312m 205u: (code 0x02000000, par16 0x0002 par64 0x0000000000000000) -> MC: exernal fault button pressed. +`
it means that the fault cable ( white cable) has been detached. could you check at the level of the bottom of the robot if you can find the white cables they all gather to a specific cable screw holder, perhaps the contact or one of the cable white get out or is not making contact, let me know if you can find it.

MichaelBombile, 2020-11-09:
Hi @julijenv,

Thanks for your response. 
I checked before the cables they seem ok but I did not check at the EMS level. I will check again if I can find any broken cable or connection and I will let you know. 

Best,

julijenv, 2020-12-30:
Hi @MichaelBombile ,
any news from that issue?

closed 2021-02-18 by violadelbono