robotology/mesh-support #1027: icubGrenoble01 refuses to start or shutdowns shortly after 0 rest position

EliseiFrederic, 2020-11-13:
## Brief description of the request/failure
icubGrenoble01 robot either (1) refuses to move on startup or (2) shutdowns nicely very shortly after normal calibration + normal rest position

## Detailed context
on fresh reboot, launching yarprobotinterface fails to startup the robot. Either
(1) nothing moves. Log describes calibration and then tries to move to zero position. Has timeouts and detects that zero position is not reached. No power in motor, nothing created
(2) moves and acts as expected : calibration is done up to the end. Zero positions are reached. And suddenly, the robot seems to shutdown gracely (parking) and then fails/stops (might seg fault yarprobotinterface)

Power supply sends the usual voltage. Does not go in alarm mode, stays in slave mode.
Emergency button checked as functional.

Behaviour (1) seems much more frequent.

## Accompanying material

We have logs for both conditions, posted just after

EliseiFrederic, 2020-11-13:
[log_icub_NeverMovesOnStartup.txt](https://github.com/robotology/icub-tech-support/files/5537340/log_icub_NeverMovesOnStartup.txt)

EliseiFrederic, 2020-11-13:
[log_icub_MovesThenAutoShutdowns.txt](https://github.com/robotology/icub-tech-support/files/5537347/log_icub_MovesThenAutoShutdowns.txt)

EliseiFrederic, 2020-11-13:
Before AutoPark, one can see:

All actions for action level 15 of startup phase started. Waiting for unfinished actions.
All actions for action level 15 of startup phase finished.
startup phase finished.
Interrupt received. Stopping all running threads.
interrupt1 phase starting...
Entering action level 1 of phase interrupt1
head_calibrator starting park device head_mc_wrapper

EliseiFrederic, 2020-11-13:
As an additional information:
the eyelids have recently broke their axis.
It was repaired a first time.
We suspect it broke again, but can't verify now. Should idle the related joint if calibration fails, but not the full robot

julijenv, 2020-11-13:
Hi @EliseiFrederic ,
could you please try to reperform the icubmoves but Autoshutdown log file with the error message instead of the time of every single messagees? please use the yarplogger and save the log.
thx in advance.

EliseiFrederic, 2020-11-13:
Hi @julijenv,Actually, these are the color codes from the Linux terminal... We'll not have access to the robot before Monday !Do you want me to process/remove the color codes on this file, to read it more easily ?Frederic

EliseiFrederic, 2020-11-15:
@baillyge did film the situation where the robot did the startup, then move to zero position, then suddenly auto shutdowns.
Here is the link:
[https://drive.google.com/file/d/1tSvB87ja6V6xPsgmxhHCL5ihHQ3kN1a7/view?usp=sharing]

Maybe it's linked to the failure of the eyelids, afterwards ?
I'll try tomorrow to comment out some parts in icub_all.xml to check this hypothesis.

EliseiFrederic, 2020-11-16:
Hi @julijenv,

I did several tests this morning, with or without TalkingHead/TalkingFace entries from icub_all.xml

The robot always started as expected. I did not experience any auto-park behaviour though, so the log attached here also includes what looks like a normal shutdown to me. 

This time, I used yarplogger + YARP_FORWARD_LOG_ENABLE=1 yarprobotinterface rather than copying the terminal output or redirecting the output to a file.

Cheers, Frédéric

EliseiFrederic, 2020-11-16:
file:///localdata/home/icub/yarprunlog_16_11_2020_11_21_13.log

EliseiFrederic, 2020-11-16:
I finally got one where the autopark failure occurred :
[yarprunlog_16_11_2020_11_43_25.logyarprunlog_16_11_2020_11_43_25--did_AutoPark.log](https://github.com/robotology/icub-tech-support/files/5545730/yarprunlog_16_11_2020_11_43_25.logyarprunlog_16_11_2020_11_43_25--did_AutoPark.log)

julijenv, 2020-11-16:
Hi @EliseiFrederic ,
`sharedcan [0] board 5 ABS ENCODER ERROR AXIS 0`
Loos like the torso ABS ENCODER is driving you crazy.
I would suggest you to double-check this by tryning to launch every single part of the robot independently - if you're not walking you could even totally comment the parts related to legs - just to simplify the debug.
If you succeed to see the part you are activated continue to repsond after the calibration phase it means this part is ok.
Go from one part to another.

EliseiFrederic, 2020-11-16:
Do you think it's a cause or a consequence ?  We did not have this error message in the previous logs (terminal logs)...

I tested with ALL components BUT the torso, and the auto shutdown still arrives.
[yarprunlog_16_11_2020_14_56_57--all_but_torso--still_shutdowns.log](https://github.com/robotology/icub-tech-support/files/5546727/yarprunlog_16_11_2020_14_56_57--all_but_torso--still_shutdowns.log)

I'll try with just one component, as you suggested.

EliseiFrederic, 2020-11-16:
Should I keep     <xi:include href="hardware/inertial.xml" /> from my icub_all.xml 
or can it be removed ?

violadelbono, 2020-11-16:
Hi @EliseiFrederic, 

We would like to see different log files for each part of the robot.

From icub_all.xml you can remove FOR EVERY TEST: MTX inertial sensor , cartesian, virtual analog sensor, skin, FT SENSOR (force-torque)
Example:
1) log with only head (Remove also face)
2) log with left arm (MAIS should stay)
3) right arm (same as above)
4) torso

Let us know where are the failures. Thanks!

EliseiFrederic, 2020-11-16:
Dear @julijenv 

I ran it many times this afternoon, adding components one after another. Or having all but one. 
I ran it several times with the original icub_all.xml, i.e. with all the components.
But I could not get it to fail again since the 14:56 one that I already reported here. Maybe the motors are warm ?
I will test again another fresh morning...

What I got from time to time is "sharedcan [x] board [y] FAULT OVERLOAD AXIS [z]" with various x,y, z
Sometimes too, the yarplog has a strange incomplete binary message/port name inbetween. But because of the ongoing experiment we don't run the very last version (YARP version 3.2.0)

julijenv, 2020-11-16:
Hi @EliseiFrederic ,
I wil be on skype on wednesday, is it fine with you to get in touch that day?
let me know your availability,
my skype-id: icub.support

EliseiFrederic, 2020-11-16:
Thanks @julijenv!
@baillyge (who has skype) and I would be available this wednesday from 9:30

EliseiFrederic, 2020-11-16:
Thanks @violadelbono
I did many runs today, but the only two that *failed* were reported here (and add several components)
Do you still want them ?
-rw-r--r-- 1 icub icub  39K nov.  16 15:22 yarprunlog_16_11_2020_15_22_31--left_arm_only.log
-rw-r--r-- 1 icub icub  77K nov.  16 15:29 yarprunlog_16_11_2020_15_29_05--both_arms_hands.log
-rw-r--r-- 1 icub icub 102K nov.  16 15:34 yarprunlog_16_11_2020_15_34_27--no_torso_no_legs.log
-rw-r--r-- 1 icub icub 109K nov.  16 15:40 yarprunlog_16_11_2020_15_40_08--no_legs.log
-rw-r--r-- 1 icub icub 9,3K nov.  16 16:50 yarprunlog_16_11_2020_16_50_14--no_legs.log
-rw-r--r-- 1 icub icub 192K nov.  16 16:55 yarprunlog_16_11_2020_16_54_42--no_legs_has_cartesian.log
-rw-r--r-- 1 icub icub 185K nov.  16 17:00 yarprunlog_16_11_2020_17_00_26--all.log
-rw-r--r-- 1 icub icub 224K nov.  16 17:12 yarprunlog_16_11_2020_17_12_42--all_but_legs.log

Do you want the ones with "sharedcan [x] board [y] FAULT OVERLOAD AXIS [z]" that I mentionned earlier ?

Thanks again for your help

violadelbono, 2020-11-16:
Sent us everything you have and we will take a look.
Thanks

EliseiFrederic, 2020-11-16:
[yarprunlog_16_11_2020_17_12_42--all_but_legs.log](https://github.com/robotology/icub-tech-support/files/5548577/yarprunlog_16_11_2020_17_12_42--all_but_legs.log)
[yarprunlog_16_11_2020_17_00_26--all.log](https://github.com/robotology/icub-tech-support/files/5548578/yarprunlog_16_11_2020_17_00_26--all.log)
[yarprunlog_16_11_2020_16_54_42--no_legs_has_cartesian.log](https://github.com/robotology/icub-tech-support/files/5548579/yarprunlog_16_11_2020_16_54_42--no_legs_has_cartesian.log)
[yarprunlog_16_11_2020_16_50_14--no_legs.log](https://github.com/robotology/icub-tech-support/files/5548580/yarprunlog_16_11_2020_16_50_14--no_legs.log)
[yarprunlog_16_11_2020_15_40_08--no_legs.log](https://github.com/robotology/icub-tech-support/files/5548581/yarprunlog_16_11_2020_15_40_08--no_legs.log)
[yarprunlog_16_11_2020_15_34_27--no_torso_no_legs.log](https://github.com/robotology/icub-tech-support/files/5548582/yarprunlog_16_11_2020_15_34_27--no_torso_no_legs.log)
[yarprunlog_16_11_2020_15_29_05--both_arms_hands.log](https://github.com/robotology/icub-tech-support/files/5548583/yarprunlog_16_11_2020_15_29_05--both_arms_hands.log)
[yarprunlog_16_11_2020_15_22_31--left_arm_only.log](https://github.com/robotology/icub-tech-support/files/5548584/yarprunlog_16_11_2020_15_22_31--left_arm_only.log)

EliseiFrederic, 2020-11-18:
Hi @violadelbono 
TeamViewer complains about our Ubuntu/Qt version :
 teamviewer dépend de libqt5gui5 (>= 5.5) | qt56-teamviewer ; cependant :
  La version de libqt5gui5:amd64 sur le système est 5.3.2+dfsg-4+deb8u3.
  Le paquet qt56-teamviewer n'est pas installé.
 teamviewer dépend de libqt5widgets5 (>= 5.5) | qt56-teamviewer ; cependant :
  La version de libqt5widgets5:amd64 sur le système est 5.3.2+dfsg-4+deb8u3.
  Le paquet qt56-teamviewer n'est pas installé.
 teamviewer dépend de libqt5qml5 (>= 5.5) | qt56-teamviewer ; cependant :
  La version de libqt5qml5:amd64 sur le système est 5.3.2-4.
  Le paquet qt56-teamviewer n'est pas installé.
 teamviewer dépend de libqt5quick5 (>= 5.5) | qt56-teamviewer ; cependant :
  La version de libqt5quick5:amd64 sur le système est 5.3.2-4.
  Le paquet qt56-teamviewer n'est pas installé.
 teamviewer dépend de libqt5dbus5 (>= 5.5) | qt56-teamviewer ; cependant :
  La version de libqt5dbus5:amd64 sur le système est 5.3.2+dfsg-4+deb8u3.

Web site says 5.2 Qt version is supported too, but I could not find the associated .deb

EliseiFrederic, 2020-11-18:
Got a version 13 working. Please mail me at frederic.elisei@gipsa-lab.fr for the ID /pass

julijenv, 2020-11-18:
Please contact :
andrea.derito@iit.it and giorgio.metta@iit.it for the quotation you were talking about.
thx

EliseiFrederic, 2020-11-18:
Thank you Julien for your help this morning.

I've been running the robot 37 times since, and did not get the failure to appear... which does not help to refine the diagnostic.
I sometimes ran software on the robot for several minutes, with no difference in the behaviour.
More than 20 times, I used the modified calibrators/face_calib.xml where the EyeLids controler was removed.
i.e. with 
        <param name="CALIB_ORDER"> (0 2 3 4 5) </param>
instead of

while at the same time legs, skin and cartesian components were removed from icub_all.xml

The other times, complimentary setup with legs and eyelids was used, but with no failure neither.
I would like to say the default has gone but as we didn't repair anything, that would be surprising. It's probably just rare...

EliseiFrederic, 2020-11-18:
Since, I've removed the eyelids parts from the head. Our model (glued twice) seems to have 2 parts only (one beeing the gear-contacting quarter circle). One reason it did break again might be that one of screws isn't holding the gear part firmly.

EliseiFrederic, 2020-11-18:
[image]

EliseiFrederic, 2020-11-18:
We'll try to print it again (the non-gear part at least) with any STL you have for our specific talking head model.
If that doesn't fit the right size, we'll have to order a new from IIT.

Thanks again, Frédéric & Gérard

julijenv, 2020-11-19:
Hi @EliseiFrederic ,
As long as you can run your experiment I would dare say that it is the most important thing for you so in some way we could be satisfy of the result - it is not perfect but fair enough to say that at leaast the robot is usable that way.
For the quotation could you please try to send it to marco.maggiali@iit.it instead of giorgio.metta@iit.it because the hierarchy inside icub facility has been changed since giorgio is now Scientific Director.
So please consider sending your request for quotation to both andrea.derito@iit.it and marco.maggiali@iit.it
best regards.

EliseiFrederic, 2020-11-19:
Hi @julijenv 
Time will say if it's enough to hide the autopark failure. Hopefully !

One thing we experience often with the modified icub_all.xml (keeping only arms+hand, torso, head and face without eyelids) 
is the final park of left hand that scrambles the fingers (at low speed, with limited power):
[image]

We used to experience this with our normal robot at shutdown sometimes when using iKin control for hand gesture. But  with the modified icub_all this is more systematic (and iKin is not used, the component is even removed from the icub_all.xml)

EliseiFrederic, 2020-11-19:
Thanks for the e-mail info, @baillyge will use it if/when needed, for the eyelids or the check-up.

julijenv, 2020-11-19:
here is the STL of the eyelids for your version

[rc_iit_023_p_001.zip](https://github.com/robotology/icub-tech-support/files/5566476/rc_iit_023_p_001.zip)

EliseiFrederic, 2020-11-19:
If that helps, here is the log (no colour code) of the sessions that ended with scrambled fingers:
[log_2020_11_19_10_48--ScramblingFingers--NormalShutdown.txt](https://github.com/robotology/icub-tech-support/files/5566492/log_2020_11_19_10_48--ScramblingFingers--NormalShutdown.txt)

EliseiFrederic, 2021-04-16:
Today, we changed the 0B9 MC4 board on the robot (see https://github.com/robotology/icub-tech-support/issues/1117)
It seems that it lets the "scrambled finger on shutdown" issue disappear.
After the completion of the ongoing social interaction experiments campaign, I'll reactivate all parts (skin, legs...) I wouldn't be surprised if the random shutdown failure disappeared altogether ;-)

EliseiFrederic, 2021-06-07:
No problem since we changed the MCU board.
Thanks again !

closed 2021-06-07 by EliseiFrederic