robotology/mesh-support #1030: Right hand middle finger distal joint cable is broken

kouroshD, 2020-11-19:
## Brief description of the request/failure

The  middle finger distal joint cable of the right hand is broken.

## Detailed context

Thanks to @julijenv  fixes last night, I managed to do a test I needed for collecting data to see the relationship between the motor currents and external forces acting on the finger. Unfortunately, I broke the right middle finger distal joint cable in this process (sorry @julijenv :(  ). 

## Accompanying material

[image]
[image]
[image]

violadelbono, 2020-11-19:
Hi @kouroshD 

The finger is fixed.

| IITCODE  |  QUANTITY | DESCRIPTION  |ALIAS   | 
|---|---|---|---|
| 2391  | 50cm  | Stainless steel micro-cable, cable construction 7x19mm, nom. diameter 0.45mm, outer diameter 0.61mm, Polyamide coated, AISI 316 - CARL STAHL GMBH  | U7194561  |   

**PS.** I noticed some problems when launching the robot interface -I recalibrated left arm wrist (j4 j5), fingers abduction (j7) and little fingers (j15) because they were out of range.

Now all the joints start correctly (no HW fault)

kouroshD, 2020-11-19:
Hi @violadelbono  Thank you a lot for taking care of the broken cable of the finger. Feel free to close the issue.

closed 2020-11-19 by violadelbono