robotology/mesh-support #1031: Failure to recognize left arm CAN board

pablin992, 2020-11-19:
## Brief description of the request/failure

'iCubSheffield01'
Ran `yarprobotinterface` with icub repositories at v1.12.1 and yarp v3.1.2.
Encountered an issue with CAN Discovery for board 10.0.1.2 - eft_arm-eb2-j4_15 : 

## Detailed context
log from `yarprobotinterface`

`[ERROR] from BOARD 10.0.1.2 (left_arm-eb2-j4_15), src LOCAL, adr 0, time 971s 420m 415u: (code 0x0500000b, par16 0x000e par64 0x0000000001000000) -> CFG: EOtheMAIS cannot be configured. can discovery fails. can address in p16, prot and vers in p64 lower 32 bits + .`

`[ERROR] from BOARD 10.0.1.2 (left_arm-eb2-j4_15) @ 971s 420m 561u: CAN discovery after 2999 ms has detected 1 missing eobrd_mais boards in CAN1:`

`[ERROR]1 of 1: missing eobrd_mais BOARD 10.0.1.2:CAN1:14`

`[ERROR]embObjMais::open() has an error in call of ethResources::serviceVerifyActivate() for BOARD left_arm-eb2-j4_15 IP 10.0.1.2 
yarpdev: ***ERROR*** driver <embObjMais> was found but could not open`

## Accompanying material

videos from left and right boards:
https://drive.google.com/drive/folders/1H73F85sO_TXgyZthR3w7C2mNt-A3aoxR?usp=sharing
[log.txt](https://github.com/robotology/icub-tech-support/files/5566678/log.txt)

Since the next action you would probably recommend is to check the firmware updater, could you please give me detailed instructions on how to do it? The link for checking firmware updater seems to be broken https://github.com/robotology/icub-firmware-build/blob/master/ETH/FirmwareUpdater.readme.maintenance.of.eth.robots.txt

julijenv, 2020-11-19:
Dear @pablin992 ,
the file you are looking for is in this link.
https://github.com/robotology/icub-firmware-build/blob/master/docs/FirmwareUpdater.readme.maintenance.of.eth.robots.txt

As a summary.:
open a teminal -X on the head (icub-head or pc104 depends which version you have even if I do believe it is icub-head as you are ETH based)
1- launch FirmwareUpdater
2- Click on discover on the ETH line
3- select the board on which is attached the MAIS (in your case it is 10.0.1.2) AND put in maintaeance mode
4- reclick on the line for the EMS (10.0.1.2) and do discover
5- see if you got the MAIS board detected.
let ,me know

pablin992, 2020-11-19:
Dear @julijenv,
Thanks for your response, this is what I get from FirmwareUpdater:
[image]

julijenv, 2020-11-19:
@Dear @pablin992 ,
Just as a matter of symmetry could you do the same procedure for board 10.0.1.3 (it is the right arm).
afterwards try to see if the MAIS board  behave the same from a brand new trn ON of the switch button, make a video.
I would also need an entire log file please, thx in advace

pablin992, 2020-11-19:
Thanks for your answer, the log file is attached in the issue description, maybe it got visually mergerd with the video link. 
However, here is the log: [log.txt](https://github.com/robotology/icub-tech-support/files/5568235/log.txt)
As I'm not with the robot right now, I will try to do the suggested procedure tomorrow.

pablin992, 2020-11-23:
Dear @julijenv,

These are the screenshots of Firmwareupdater
[image]
[image]

When I checked the left hand MAIS board, I figured out that a cable was disconnected from the board: 
https://drive.google.com/file/d/1IA-zy9Bczmb660UOndSXs0DEPRpJSsj9/view?usp=sharing

The Issue may be due to that disconnection. However, when I run yarprobotinterface, a red led starts blinking in the right arm. as you can see in the video:

https://drive.google.com/file/d/1I9QJs2Vw5XkKqhEGtpRHY1RZrmcDcgE6/view?usp=sharing

I also recorded the requested procedure for the ON sequence:

https://drive.google.com/file/d/1I9QJs2Vw5XkKqhEGtpRHY1RZrmcDcgE6/view?usp=sharing

Thanks in advance

julijenv, 2020-11-23:
HI @pablin992 ,
THe result of the different videos and photos need to be clarify but we touche the end for sure.
TO me having found the cable disconnected, should have solved the issue - the last videos show that the MAIS boards start the same way.
The last thing, could you try to launch the robot and record it.
and finally, use Firmwarepdater , please show the PrintScreen of the 10.0.1.2 and 10.0.1.4 - you should see beneath every board listed one MAIS board.
 could you confirm?
as long as you cant see the MAIS beneath 10.0.1.2 it means the cable you thougt you restored is not enough. please consider make a continuity test in between the connector on which you found the wire broekn and the connector on the MAIS board itself.
let me knwo

pablin992, 2020-11-27:
Dear @julijenv,

connecting the cable solved the issue. Thanks!

closed 2020-11-27 by pablin992