robotology/mesh-support #1032: iCubErzelli01: Left hand calibration issue

ddetommaso, 2020-11-19:
## Brief description of the request/failure
The calibration of the left hand does not complete. The left hand remains closed. 

## Detailed context
In yarpmotorgui the JOINT 15 it is in NOT CONFIGURED state, while the encoder value indicates about -11.0

## Accompanying material
[log_icub-head_yarprobotinterface_2825.txt](https://github.com/robotology/icub-tech-support/files/5568540/log_icub-head_yarprobotinterface_2825.txt)

julijenv, 2020-11-20:
Hi @ddetommaso ,
It seems that either as I explained to you last time we rebuild the new calibration of the last joint of the hand (meaning closing totally the fingers as the rest of the jlints) and consequently the joint should be correctly caibrated, or recalibrateing the last joint to say -11 = 0.
Last possibility just close a bit the last fingers joint when it calibrate in the meanwhile and the caliration of the entire hand should perform normally.
hope it helps

ddetommaso, 2020-11-24:
Hi @julijenv 

closing a bit the last fingers solved the issue. Thank you

violadelbono, 2021-01-18:
Hi @ddetommaso,

In order to avoid this problem we changed the initial calibration of the hands by adding a closing of the ring and pinkie finger (joint 15).

FYI these are the files that have been modified:

- `calibrator/right_arm-calib.xml` and `calibrator/left_arm-calib.xml` :  parameter calibration2 from 3360 to 10000 and calibration3 parameter: from -1 to 1
- ` hardware/mechanicals/right_arm-eb29-j12_15-mec.xml` and `hardware/mechanicals/left_arm-eb26-j12_15-mec.xml` : parameters RotorPosMin e RotorPosMax swapped and multiplied by  -1

closed 2021-01-18 by violadelbono