robotology/mesh-support #1033: Oil in the intersection of iCub 3 legs

DanielePucci, 2020-11-20:
## Brief description of the request/failure
After preliminary operations with iCub 3, we noticed some oil at the intersection of its legs 

## Accompanying material

[image]

julijenv, 2020-11-20:
Hi @DanielePucci ,
to be more precise , The grease/oil is coming out of the right hip flexion motor. I will inspect and as kthe builder for help

marcellosavoldi, 2020-11-20:
Hi guys,
We known the problem .... this is due after a long period of inactivity, the grease inside the harmonic drive produces oil.
For now we can clean the drop and hope that after a good using the problem solved ( the grease should be mixed again ).
Keep attention that the electrical component are cleaned.
keep me tuned about the problem.

cc: @fiorisi